hifiberry/hifiberry-os #213: Spotify shows up but can’t connect

agentff6600, 2021-01-07:
DO NOT remove the blocks below, but fill these with the requested data. Incomplete bug reports will be ignored!
You should remove this comment before posting the bug report.

**Describe the bug**
Spotify connect shows up in the player but unable to connect to it. On other occasions the spotify player tries to connect and tries to play the track. after a few seconds, connection is dropped.

**HiFiBerryOS version**
20201213

**HiFiBerry sound card**
DAC+/amp2

**To Reproduce**
In general, device shows up bit can’t be connect.

Before this happened:
1. Spotify Player: select a device (spotify connect)
2. See player connected 
3. Song playing, no audio 
4. Disconnected from device

**Expected behavior**
Connecting and playing spotify songs through spotify connect.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Browser (if applicable)**
 - OS:  iOS
 - Browser safari
 - Version  22

**Additional context**
Add any other context about the problem here.

hifiberry, 2021-01-08:
Did you enable the equalizer?

agentff6600, 2021-01-12:
@hifiberry yes.

Turned it off and restarted but still can’t access the connect client from the spotify player.

hifiberry, 2021-01-15:
Check the output of 

journalctl -u spotify 

when it drops the connection. I don't know if we can do anything about it, but at least it might give some ideas what's going on.

tysonnorris, 2021-01-18:
I experienced the same on a fresh install of hifiberryos on a pi4 with AMP2. Airplay works fine, just not spotify connect. Same symptoms: device is listed, indicates connected, no audio during playback.  I did enable equalizer.

I have a separate working pi3 using librespot (not on hifiberryos) that seems to work more consistently, I will see if there are some differences in versions (spotifyd uses librespot) that I can make work.

tysonnorris, 2021-01-19:
here's the full log from when connect occurs, then play, then no audio - notice the ALSA error 
```
Jan 18 16:17:08 hifiberry vollibrespot[4486]: [Vollibrespot] : Connecting to AP "guc3-accesspoint-e-g486.ap.spotify.com:443"
Jan 18 16:17:09 hifiberry vollibrespot[4486]: [Vollibrespot] : Authenticated as "noizenheimer" !
Jan 18 16:17:09 hifiberry vollibrespot[4486]: [Vollibrespot] : Setting up new mixer: card:hw:0 mixer:Digital index:0
Jan 18 16:17:09 hifiberry vollibrespot[4486]: [Vollibrespot] : Alsa Mixer info min: 0 (MilliBel(-9999999)[dB]) -- max: 207 (MilliBel(0)[dB]) HW: true
Jan 18 16:17:09 hifiberry vollibrespot[4486]: [Vollibrespot] : Using alsa sink
Jan 18 16:17:09 hifiberry raat_app[602]: [0001447] [t3058869328] 277094.939 INFO [volume/alsa] changed!
Jan 18 16:17:09 hifiberry raat_app[602]: [0001448] [t3058869328] 277094.939 TRACE [volume/alsa] read values: min=-100.000000, max=0.000000, value=-18.000000, mute=0
Jan 18 16:17:09 hifiberry vollibrespot[4486]: [Vollibrespot] : Metadata pipe established
Jan 18 16:17:09 hifiberry vollibrespot[4486]: [Vollibrespot] : Country: "US"
Jan 18 16:17:09 hifiberry vollibrespot[4486]: [Vollibrespot] : Event: Volume { volume_to_mixer: 32767 }
Jan 18 16:17:09 hifiberry python3[342]: ERROR:root:Don't know what to do with vollibrespot v0.2.2
Jan 18 16:17:09 hifiberry beocreate2[685]: Volume received from AudioControl: 83 %.
Jan 18 16:17:09 hifiberry vollibrespot[4486]: [Vollibrespot] : Fetching autoplay context uri
Jan 18 16:17:09 hifiberry vollibrespot[4486]: [Vollibrespot] : Event: SessionActive { became_active_at: 1611015429762 }
Jan 18 16:17:09 hifiberry vollibrespot[4486]: [Vollibrespot] : SessionActive!
Jan 18 16:17:09 hifiberry vollibrespot[4486]: [Vollibrespot] : Event: PlaybackStarted { track_id: SpotifyId { id: 128060244776819574859106233703860368367, audio_type: Track } }
Jan 18 16:17:09 hifiberry vollibrespot[4486]: [Vollibrespot] : Event: PlaybackLoading { track_id: SpotifyId { id: 128060244776819574859106233703860368367, audio_type: Track } }
Jan 18 16:17:09 hifiberry vollibrespot[4486]: [Vollibrespot] : Autoplay uri resolved to <"spotify:station:playlist:4n1KtzMJUvh8enJhXWInfb">
Jan 18 16:17:09 hifiberry vollibrespot[4486]: [Vollibrespot] : Resolving uri "spotify:station:playlist:4n1KtzMJUvh8enJhXWInfb"
Jan 18 16:17:09 hifiberry vollibrespot[4486]: [Vollibrespot] : Event: GotToken { token: Token { access_token: "<removed>", expires_in: 3600, token_type: "Bearer", scope: ["streaming", "user-read-playback-state", "user-modify-playback-state", "user-read-currently-playing", "user-read-private"] } }
Jan 18 16:17:09 hifiberry vollibrespot[4486]: [Vollibrespot] : Loading <Burning Down the House> with Spotify URI <spotify:track:2VNfJpwdEQBLyXajaa6LWT>
Jan 18 16:17:10 hifiberry beocreate2[685]: Source 'spotify' has activated.
Jan 18 16:17:10 hifiberry vollibrespot[4486]: [Vollibrespot] : Resolved 50 tracks from <"spotify:playlist:4n1KtzMJUvh8enJhXWInfb">
Jan 18 16:17:11 hifiberry vollibrespot[4486]: [Vollibrespot] : <Burning Down the House> (241920 ms) loaded
Jan 18 16:17:11 hifiberry vollibrespot[4486]: [Vollibrespot] : Event: PlaybackStopped { track_id: SpotifyId { id: 128060244776819574859106233703860368367, audio_type: Track } }
Jan 18 16:17:11 hifiberry beocreate2[685]: Source 'spotify' has deactivated.
Jan 18 16:17:12 hifiberry python3[342]: 127.0.0.1 - - [18/Jan/2021 16:17:12] "GET /api/player/status HTTP/1.1" 200 738
Jan 18 16:17:23 hifiberry beocreate2[685]: Source 'spotify' has activated.
Jan 18 16:17:31 hifiberry beocreate2[685]: Checking MPD database update status...
Jan 18 16:17:53 hifiberry vollibrespot[4486]: [Vollibrespot] : Event: Play { track_id: SpotifyId { id: 128060244776819574859106233703860368367, audio_type: Track }, position_ms: 10744 }
Jan 18 16:17:53 hifiberry python3[342]: WARNING:root:don't know how to handle {'state': {'status': 'play'}}
Jan 18 16:17:53 hifiberry /bin/cat[18216]: 20210118-161753 20210118-161753 pause-all vollibrespot
Jan 18 16:17:53 hifiberry vollibrespot[4486]: [Vollibrespot] : stopped other players
Jan 18 16:17:53 hifiberry vollibrespot[4486]: [Vollibrespot] : Alsa error PCM open ALSA function 'snd_pcm_open' failed with error 'EPERM: Operation not permitted'
Jan 18 16:17:53 hifiberry vollibrespot[4486]: [Vollibrespot] : Could not start audio: Alsa error: PCM open failed
Jan 18 16:17:53 hifiberry kernel: pcm512x 1-004d: No SCLK, using BCLK: -2
Jan 18 16:17:53 hifiberry python3[342]: ERROR:root:{'album_id': '1invpprZoPxP2H3CfGgSfJ', 'album_name': 'Burning Down the House / I Get Wild / Wild Gravity', 'albumartId': ['ab67616d00001e02ee320a29df2f19c3d900b83d', 'ab67616d00004851ee320a29df2f19c3d900b83d', 'ab67616d0000b273ee320a29df2f19c3d900b83d'], 'artist_id': ['2x9SpqnPi8rlE9pjHBwmSC'], 'artist_name': ['Talking Heads'], 'duration_ms': 241920, 'position_ms': 10744, 'track_id': '2VNfJpwdEQBLyXajaa6LWT', 'track_name': 'Burning Down the House'}
Jan 18 16:17:53 hifiberry /bin/cat[18238]: vollibrespot (4486) requested pause, but is already using sound card, ignoring
Jan 18 16:17:53 hifiberry vollibrespot[4486]: [Vollibrespot] : stopped other players
Jan 18 16:17:53 hifiberry vollibrespot[4486]: [Vollibrespot] : Alsa error PCM open ALSA function 'snd_pcm_open' failed with error 'EBUSY: Device or resource busy'
Jan 18 16:17:53 hifiberry vollibrespot[4486]: [Vollibrespot] : Could not start audio: Alsa error: PCM open failed
[... cut ...]
Jan 18 16:18:13 hifiberry restore-volume[18288]:   Capabilities: volume
Jan 18 16:18:13 hifiberry restore-volume[18288]:   Playback channels: Front Left - Front Right
Jan 18 16:18:13 hifiberry restore-volume[18288]:   Capture channels: Front Left - Front Right
Jan 18 16:18:13 hifiberry restore-volume[18288]:   Limits: 0 - 255
Jan 18 16:18:13 hifiberry restore-volume[18288]:   Front Left: 255 [100%]
Jan 18 16:18:13 hifiberry restore-volume[18288]:   Front Right: 255 [100%]
Jan 18 16:18:13 hifiberry systemd[1]: Started Vollibrespot.
```

tysonnorris, 2021-01-19:
FWIW my working system on pi3 with raspbian lite is using a version of librespot with raspotify 0.16.0
```
Jan 19 02:05:20 raspberrypi librespot[13184]: [2021-01-19T02:05:20Z INFO  librespot] librespot (raspotify v0.16.0) 0adb851 (2020-12-15). Built on 2020-12-30. Build ID: KLdkAvgt
```

I'm not sure this helps - I'm not familiar with vollibrespot

tysonnorris, 2021-01-19:
When I disabled airplay and equaliser (not sure if one or the other or both was the issue), hifiberry + spotifyconnect started working again.

hifiberry, 2021-01-19:
Spotify isn't compatible with the EQ. We'll probably disable this module in one of the next releases.

closed 2021-01-20 by hifiberry

tysonnorris, 2021-01-23:
@hifiberry I was able to get spotify + eq working by adding `controls "/root/.alsaequal.bin"` to the `pcm.equal` config e.g.
```
pcm.equal {
  type equal;
  slave.pcm "plug:softvol";
  controls "/root/.alsaequal.bin"
}
```
Maybe there is something in [this patch](https://github.com/hifiberry/alsaequal/commit/23f7609d63faf4f502d5fc441affba254c9bdabd) that is preventing the default value for `controls` to be used?