hifiberry/hifiberry-os #214: Cracks ans pops / dac+dsp / 20201213 hifiberryos

melomane63, 2021-01-09:
Describe the bug 
 1) two loud crack on startup /  with DSP-V10 NO CRACKS 
 2) loud pop on stop playing /  only if Spdif is active and newer played / same behaviour with DSP-V10

HiFiBerryOS version
20201213 “DAC+ DSP Universal Version 12”

HiFiBerry sound card
DAC+DSP

Expected behaviour
No cracks and pops

schwachstelle, 2021-01-11:
On my DAC+ DSP (program version 12) with optical input activated I notice a pop sound in the following situations:
- changing song on Spotify (automatic change within a playlist)
- ~7 seconds after pausing a song on Spotify
There is no pop sound when optical input is disabled.

hifiberry, 2021-01-15:
The spotify backend doesn't really handle skipping and stopping well. Do you see this also with other sources? If only Spotify is affected, there is probably nothing we can do about it.

melomane63, 2021-01-15:
all sources product louder pop on stop playing, not only spotify. 
this pop appear ONLY is spdif is active, the high the spdif volume, high the pop. 
if you feed signal to spdif and after stop feed signal, NO MORE pop. 

i will try in the afternoon if there is the same problem with precedend hifiberryos

hifiberry, 2021-01-15:
That looks like some DC offset from SPDIF.

melomane63, 2021-01-15:
ok, i have the same behaviour with 20201101 os and DSP 10.

melomane63, 2021-01-15:
pops also in 20201202 ( spdif on and newer played ) 

No cracks on startup

kevintully, 2021-02-05:
I'm also experiencing crackling and pops on the SPDIF input on my BeoCreate board after upgrading to HifiBerryOS 20201213.

At first, I was missing the optical source entirely, but after following the thread for issue #197, I was able to select the SPDIF source. It was at that point that the crackling began.

I use a DAC2 Pro elsewhere in the house, also using SPDIF, and it has been performing without issue after the 20201213 upgrade.

I've tried to isolate the issue to the BeoCreate board by using known-good SPDIF cables and used a known-good SPDIF source that works without issue on the DAC2 Pro. 

Becuase the update works great on my DAC-only boards, I'm wondering if this is perhaps related to the DSP or speaker profiles? The BeoCreate I'm using is in a Beovox CX-100 stereo pair, and along with the crackling/pop sounds, the equalizer profile for the CX-100 also seemed to just completely eliminate the high end and was only resolved by disabling the profile (but that's for a separate issue thread!)

I'm happy to help troubleshoot or pull logs if it would help. I'll keep exploring in the meantime.

melomane63, 2021-02-14:
With 20210203 and dsp12 
loud crack on start-up are gone

  1. two loud crack on start-up /  with DSP-V10 NO CRACKS
  2. loud pop on stop playing /  only if Spdif is active and newer played gone also

BUT now i have  crack on change song... level isn't very big, but it is noticeable.