hifiberry/hifiberry-os #216: Add Interaction 'Play Favorite Radio'

greewoo, 2021-01-14:
DO NOT remove the blocks below, but fill these with the requested data. Incomplete feature requests will be closed.
You should remove this comment before posting the feature request.

**Is your feature request related to a problem? Please describe.**
No auto-play for Radio, when Startup HiFiBerryOS.

**Describe the solution you'd like**
Add Interaction to play Favorite Radio (1, 2, 3, ..., n) to 'Action'

**Describe alternatives you've considered**
Start playing Radio Favorite 1 after Startup HiFiBerryOS

hifiberry, 2021-01-15:
We're working on documentation on the Interact modules that should enable contributors to create additional interactions.

LordEidi, 2021-12-31:
I had the same problem and solved it with a hack. See here, in case you are interested: 

`https://github.com/LordEidi/hifiberryos_hacks`

But I would prefer to do it in a more integrated way. Is there any good intro to the Interact modules, how to write them, how to extend the HifiberryOS? I understood, that the server is nodejs based. And I found a few UI components. But nothing re integration and extension with new functionality...