hifiberry/hifiberry-os #219: Spotify Connect song title and artwork not updating

eproxus, 2021-01-16:
**Describe the bug**
After a while the Spotify Connect feature stops updating the artist/song titles and the album artwork. Playback still works fine. This persists between separate playback sessions. A restart temporarily solves the problem. 

**HiFiBerryOS version**
20201213

**HiFiBerry sound card**
DAC2 HD

**To Reproduce**
Steps to reproduce the behavior:
1. Play music via Spotify Connect
2. Wait a while (unknown amount of time)
3. Observe song information and artwork stuck at some previously played track

**Expected behavior**
Song info and artwork always up to date with what the daemon is currently playing.

**Browser (if applicable)**
 - OS: iOS 14.3
 - Browser: Safari

**Additional context**
Tested both Never and If Higher Resolution settings for downloading artwork, but error shows for both.

hifiberry, 2021-01-20:
To better understand if this is a bug in the frontend or the backend, please run the following command when you see this happening again:

curl 127.0.0.1:81/api/player/status

This gives us an internal state table.

eproxus, 2021-01-28:
```
# curl 127.0.0.1:81/api/player/status{"players": [{"name": "mpd", "state": "stopped", "artist": null, "title": null, "supported_commands": ["Next", "Previous", "Pause", "PlayPause", "Stop", "Play", "Seek", "Random", "RandomOff", "RepeatAll", "RepeatOff"]}, {"name": "spotify", "state": "paused", "artist": "Just Brothers", "title": "Sliced Tomatoes", "supported_commands": ["Next", "Previous", "Pause", "PlayPause", "Play"]}, {"name": "ShairportSync", "state": "playing", "artist": "Gigi Masin", "title": "The Word Love - Original Mix \u2022 Gigi Masin", "supported_commands": ["stop", "pause", "next", "previous", "play"]}], "last_updated": "2021-01-28 17:19:41.297876"}
```

Current state is:
* Cannot pause/skip in the web UI (volume works)
* Web UI shows the song "Sliced Tomatoes" (see screenshot)
* Spotify on iOS shows whatever song is actually playing (playing works fine)
* The player status above shows the song "The Word Love [...]"
* The last updated timestamp is ~3 hours old

[image]

hifiberry, 2022-10-10:
This was probably caused by a bug in audiocontrol. Do you still see this in the current version?