hifiberry/hifiberry-os #221: Volume level jumps to max at each source connection

RavanH, 2021-01-18:
**Describe the bug**
When reconnecting my phone via bluetooth (a known device, connected before then deconnected by walking out of the room for example) the volume level set via the HifiBerryOS web interface jumps to 100% instead of sticking with the previously set level.

**HiFiBerryOS version**
20201213

**HiFiBerry sound card**
AMP2

**To Reproduce**
Steps to reproduce the behavior:
1. Connect a phone via bluetooth to the HifiBerry and play some music
2. Go to the HifiBerry web interface and reduce the sound level on the bottom player bar
3. Deactivate bluetooth on the phone
4. Reactivate bluetooth and let the phone reconnect and start playing music via HifiBerry again

**Expected behavior**
Same sound level after reconnexion as before de-connexion

**Browser (if applicable)**
 - OS: Ubuntu 20.10
 - Browser Chrome
 - Version 87

hifiberry, 2021-01-18:
Unfortunately there isn't really something we can do about this. BT audio/volume support is handled by a patch as the BT audio implementation still doesn't support volume control. 
Unfortunately there is no progress with an "official" version:
https://github.com/Arkq/bluez-alsa/pull/321

hifiberry, 2021-01-18:
In general, I would recommend to stay away from BT audio whenever possible as this usually creates the most problems.

RavanH, 2021-01-18:
Hi, please be aware that this does not apply to bluetooth alone. I get the volume level jump back to 100% when switching Radio channels. Depending on the channel (and the their current sound track) that can bring quite a scare, I can tell you :/

hifiberry, 2021-01-18:
That's very strange. I can't see this here. If you completely disable BT on the Pi you see that the volume changes just by starting a new radio channel?

RavanH, 2021-01-18:
Ah yes, you are right. This does not happen anymore after a reboot and there is no BT connexion made (yet)...

hifiberry, 2021-01-18:
Best workaround we might be able to do is a way to disable BT volume control completely.

RavanH, 2021-01-18:
I'm not sure (or understand) why BT should be able to control audio volume level at all...

hifiberry, 2021-01-19:
Because that's what most people would expect when connecting a BT speaker.

RavanH, 2021-01-19:
Ah... ok, so that's for an output device. Would it be possible to disable volume control for input devices only?

hifiberry, 2021-01-19:
From the BT point of view your HiFiBerry system IS an output device.

closed 2022-05-17 by hifiberry

leonka10, 2023-06-30:
I have the same problem here. It seems the volume gets set to 100% when I press play in the Spotify App on iPhone. I also noticed turning down the volume with a rotary control wired to the Raspi does lower the volume but does not effect the volume control bar on the iPhone.