hifiberry/hifiberry-os #223: Feature Request: Integrated Snapcast Server

gpatkinson, 2021-01-20:
Hi, I've tried many of the competing projects out there (most recently was using volumio) and I really like hifiberry! Thanks for all your hard work.

I see that HifiberryOS supports connecting to a snapcast server ... what would be really great is if there was one integrated into HifiberryOS so that we can synchronously stream to other rooms. Is this on the roadmap? Thanks for your consideration.

hifiberry, 2021-01-20:
Not yet on the roadmap. As HiFiBerryOS is designed as a player and not as a server, we'll probably won't look into this soon. However, using Docker it might be possible to implement this. Looking forward to contributions...

BTW: If you want us to leave this open, please use the Feature request template that we provide.

closed 2021-01-20 by hifiberry

gpatkinson, 2021-01-22:
Sorry about not using the proper template, noted for next time.

Yeah, I'll see if I can put together an alpine-based image that runs snapcast. Will make a pull request if I can get it working. Thanks again for this great software!