hifiberry/hifiberry-os #225: AirPlay 2 Support

gustavneustadt, 2021-01-22:
**Is your feature request related to a problem? Please describe.**
Currently no AirPlay 2 support.

**Describe the solution you'd like**
Support for AirPlay 2.

**Describe alternatives you've considered**
None.

**Additional context**
Seems like the folks at [shairport-sync](https://github.com/mikebrady/shairport-sync/issues/535) got AirPlay 2 reverse engineered. Looking forward for a implementation into HifiBerry OS. Maybe you were already aware of that.

hifiberry, 2021-01-24:
When shairport-sync supports this, we'll most likely integrate this. As this is not yet ready, we won't dig into this now.

neiltron, 2021-07-12:
Looks like [support has landed](https://github.com/mikebrady/shairport-sync/issues/535#issuecomment-876468388) in shairplay-sync's `develop` branch 🎉🎉🎉

hifiberry, 2021-07-12:
Yes, but it's still very rough. No cover art support and it won't run on the Pi Zero and Pi2. This means quite some work to package it, even if it works. We'll look into it, but it might still take some time until we feel it's ready...

hifiberry, 2021-07-22:
Now we have the same feature request 2 times :(
https://github.com/hifiberry/hifiberry-os/issues/275

hifiberry, 2021-07-23:
If somebody wants to try it. There's now an experimental release with Airplay 2 support. Seems to work here. Only for Pi3/4!

/opt/hifiberry/bin/update --experimental

Smith8154, 2021-07-24:
I’ve been testing it for a few hours now and it is working great! Audio is perfectly in sync when playing back to multiple devices, it is quick to connect and disconnect, and playing and pausing is institutions. I have a Pi 4 and CPU usage seems to be hovering around 25% while playing back. Only issue is no album artwork, but I believe that was listed as not working on the shairport-sync page.

meijkl, 2021-08-01:
I've also installed the experimental version (HW: RPi 3b + DAC2 Pro) three days ago and so far it is working very well!

P.S.: I saw some (minor?) commits after the experimental release - is a newer version already available? Thanks!

hifiberry, 2021-08-02:
No new version, we don't create releases from every commit.

meijkl, 2021-08-02:
Thanks for the info!
Looking forward to upcoming releases with new features :-)

neiltron, 2021-08-04:
Working nicely for me on two raspis aside from some occasional disconnects when pausing audio. Thanks for adding it to the experimental release.

YhQb6, 2021-08-05:
how do i upgrade to the experimental version to get airplay 2 support?

YhQb6, 2021-08-05:
I figured it out

YhQb6, 2021-08-05:
I installed the experimental version what else do i have to do to enable airplay 2

hifiberry, 2021-08-05:
Nothing

YhQb6, 2021-08-05:
how do I install this experimental version? I thought I had it figured out but didn't. I am on 20210804

YhQb6, 2021-08-05:
I had to have the second Airplay 2 enabled device running.

mungler, 2021-11-04:
Can i suggest it might be nice to have the old shairport as an option? I've been trialling a setup with six Pi3s each with and Amp2, and while its great when it works, and I can get whole-home, in-sync audio, it's a little unstable at times and I'm getting complaints from family members about disconnections or things just not working. 

I'm now going to try another approach with Airfoil running on a Mac mini as a central hub, streaming in sync to the Pis, but for this, Airplay1 is sufficient as Airfoil deals with synchronisation, and may be more stable (and allow album art etc). 

If it was an option to have both versions installed on hifiberryos, on capable hardware, as two separate sources, that would be ideal.

hifiberry, 2021-11-04:
Unfortunately we can't do this. You would either need to stick to an older HiFiBerryOS version or create your own build.

Yashinder, 2021-11-19:
I solved this problem when changed cpu power governor from powersave to performance. This method also helps with pops between tracks.

mungler, 2021-11-19:
That sounds awesome @Yashinder - how do I go about doing that?

mungler, 2021-11-19:
So I found the file `scaling_governor` at /sys/devices/system/cpu/cpufreq/policy0 after ssh-ing into a Pi3 running HifiberryOS. If I modify the contents from `powersave` to `performance` will this 'stick' after a reboot? I remember reading something about having to build your own HifiberryOS image to get stuff to stick.

Alternatively is there a startup scripts location I can use to modify on boot?

Any guidance appreciated @hifiberry :)

Yashinder, 2021-11-19:
Create sh script:
`touch /etc/performance.shchmodu+r+x`

Open this:
`nano /etc/performance.sh`

Here is the code of script:
```
#!/bin/bash
echo performance > /sys/devices/system/cpu/cpu0/cpufreq/scaling_governor
```

Create your systemd service:
`touch /etc/systemd/system/power-script.servicechmod664`

Open this:
`nano /etc/systemd/system/power-script.service`

Here is the text of code:
```
[Unit]
Description=Disable energy saver
After=network.target

[Service]
Type=oneshot
User=root
ExecStart=/etc/test.sh

[Install]
WantedBy=multi-user.target
```

Reload systemd:
`systemctl daemon-reload`

Try to start your service:
`systemctl start power-script.service`

If all works fine and you are not see any warning messages, enable the service for each boot of Hifiberry OS:
`systemctl enable power-script.service`

You must see:
`Created symlink from /etc/systemd/system/multi-user.target.wants/power-script.service to /etc/systemd/system/power-script.service.`

After that reboot your device and check which power governor is in use. This works great for my device.

mungler, 2021-11-19:
Hey, thanks a lot @Yashinder !

I guess the chmod line should be separate?

Will give this a try, thanks again.

Yashinder, 2021-11-19:
> Hey, thanks a lot @Yashinder !
> 
> Will give this a try, thanks again.

Your welcome. After several tests, it is clear that the system works better, but not perfect.

> I guess the chmod line should be separate?
This is how it works in my case

Yashinder, 2021-11-20:
How I solved the problem with an Shairport short connection delay.
According to the instruction from the Shairport, you can set a timeout when starting the service.

> -t timeout | --timeout=timeout Exit play mode if the stream disappears for more than timeout seconds. When shairport-sync plays an audio stream, it startes a play session and will return a busy signal to any other sources that attempt to use it. If the audio stream disappears for longer than timeout seconds, the play session will be terminated. If you specify a timeout time of 0, shairport-sync will never signal that it is busy and will not prevent other sources from "barging in" on an existing play session. The default value is 120 seconds.

Need to ssh login to the device.
Run `nano /lib/systemd/system/shairport-sync.service`
In the line ExecStart=/bin/shairport-sync add ` -t 180` or other value what you want.
Save, exit and reboot the device. Works 100%

mungler, 2021-11-20:
@Yashinder you da man. Away from home right now but will be making these changes soon as I'm home. Thanks again!

Yashinder, 2021-11-20:
And now the right way to solve the problem of short connection timeout:
1. SSH login to the device.
2. `nano /etc/shairport-sync.conf`
3. Set the "session_timeout = " to 120 or another value in seconds that you want.
4. Save, exit, reboot and enjoy

mungler, 2021-11-22:
@Yashinder thank you once again. I've now set all 6 of my Pis up with the performance mode and a longer timeout, will see how it performs over the next few days. Much appreciated.

hifiberry, 2021-11-22:
Interesting. We'll update this in the next release.

gonewandering, 2022-10-05:
Any update on this? The one feature that would make Hifiberry usable for multiroom audio

hifiberry, 2022-10-06:
This has been updated already some time ago.

closed 2022-10-06 by hifiberry

PatrykDampc, 2022-10-16:
But how to enable this? I have most recent version of Hifiberry OS but there is no option for enabling multiroom audio nor there isn't any word on readme about AirPlay 2 support

tastyshrimps, 2022-10-16:
It just works ;-) the other devices that play need to be airplay 2 enabled,
too, of course.