hifiberry/hifiberry-os #226: Integrated Snapcast Server

gpatkinson, 2021-01-22:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

Not related to a problem/bug in software, but rather a common problem in home music set-ups: synchronizing music streams in different rooms

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

Integrate a snapcast server option into hifiberryOS

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

Running a separate snapcast server.

**Additional context**
Add any other context or screenshots about the feature request here.

Considering Docker/Container as a way to integrate.

hifiberry, 2021-01-24:
We don't plan to do this in the near guture, but we're open to contributions.

gpatkinson, 2021-01-24:
I’ve got some ideas about how to do it and will give it a try. Will make pull request if I can get it working. Thanks for the great software!

hifiberry, 2022-07-21:
If you have a pull request, please send it.