hifiberry/hifiberry-os #227: screen resolution for small displays

yvaditya, 2021-01-24:
DO NOT remove the blocks below, but fill these with the requested data. Incomplete feature requests will be closed.
You should remove this comment before posting the feature request.

**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

yvaditya, 2021-01-24:
[image]
[image]

yvaditya, 2021-01-24:
trying to build a dac with almost a coverart display, can you add a way to set resolution, and these cheap displays only show up as a 1080p display unless a resolution is pushed in, also a large ui buttons, but this could be though a HTML template. I get you want to maintain the minimal OS look and feel, but also as a os designed for diyers, most of the times these contradict each other. may be add it in the config.txt so only if people how really want it can edit it,

hifiberry, 2021-02-01:
What exactly did you edit?

yvaditya, 2021-02-03:
i added this in the config.txt, to set the resolution and refresh rate to the correct ones, 

hdmi_group=2
hdmi_mode=87
hdmi_cvt=800 480 60 6 0 0 0
hdmi_drive=1

closed 2021-02-03 by hifiberry