hifiberry/hifiberry-os #229: buildroot compile fails after error on upnpp 0.20.0 download returns 404

tysonnorris, 2021-01-25:
DO NOT remove the blocks below, but fill these with the requested data. Incomplete bug reports will be ignored!
You should remove this comment before posting the bug report.

**Describe the bug**
I'm trying to compile after build-config 3, and get an error downloading a missing file for upnpp:
```
 /usr/bin/install -c -m 644 libupnp.pc '/home/ubuntu/buildroot-3/target/usr/lib/pkgconfig'
>>> libupnpp 0.20.0 Downloading
--2021-01-25 03:43:52--  http://www.lesbonscomptes.com/upmpdcli/downloads/libupnpp-0.20.0.tar.gz
Resolving www.lesbonscomptes.com (www.lesbonscomptes.com)... 62.210.16.61
Connecting to www.lesbonscomptes.com (www.lesbonscomptes.com)|62.210.16.61|:80... connected.
HTTP request sent, awaiting response... 301 Moved Permanently
Location: https://www.lesbonscomptes.com/upmpdcli/downloads/libupnpp-0.20.0.tar.gz [following]
--2021-01-25 03:43:52--  https://www.lesbonscomptes.com/upmpdcli/downloads/libupnpp-0.20.0.tar.gz
Connecting to www.lesbonscomptes.com (www.lesbonscomptes.com)|62.210.16.61|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2021-01-25 03:43:53 ERROR 404: Not Found.

--2021-01-25 03:43:53--  http://sources.buildroot.net/libupnpp/libupnpp-0.20.0.tar.gz
Resolving sources.buildroot.net (sources.buildroot.net)... 104.26.1.37, 104.26.0.37, 172.67.72.56, ...
Connecting to sources.buildroot.net (sources.buildroot.net)|104.26.1.37|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2021-01-25 03:43:53 ERROR 404: Not Found.

--2021-01-25 03:43:53--  http://sources.buildroot.net/libupnpp-0.20.0.tar.gz
Resolving sources.buildroot.net (sources.buildroot.net)... 172.67.72.56, 104.26.0.37, 104.26.1.37, ...
Connecting to sources.buildroot.net (sources.buildroot.net)|172.67.72.56|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2021-01-25 03:43:53 ERROR 404: Not Found.

package/pkg-generic.mk:144: recipe for target '/home/ubuntu/buildroot-3/build/libupnpp-0.20.0/.stamp_downloaded' failed
make[1]: *** [/home/ubuntu/buildroot-3/build/libupnpp-0.20.0/.stamp_downloaded] Error 1
Makefile:84: recipe for target '_all' failed
make: *** [_all] Error 2
```

**HiFiBerryOS version**
building from master as of `c13f386d80c7f558cf5fa21d9ff873e4a4a62ae4`

**HiFiBerry sound card**
n/a

**To Reproduce**
Steps to reproduce the behavior:
1. 
 ``` 
./get-buildroot
./build-config 3
./compile
```

2. Wait
3. See error after 404 for `libupnpp-0.20.0.tar.gz`

Note that this file doesn't exist at any of:
* http://www.lesbonscomptes.com/upmpdcli/downloads/libupnpp-0.20.0.tar.gz
* http://sources.buildroot.net/libupnpp/libupnpp-0.20.0.tar.gz
* http://sources.buildroot.net/libupnpp-0.20.0.tar.gz

But a new version(s) exist:
* 0.20.1
* 0.20.2

I will test out an update to `buildroot-2020.11.patch` to see if compile will complete.

**Expected behavior**
No compile errors

**Screenshots**
n/a

**Browser (if applicable)**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Additional context**
Building on ubuntu 18.04 AMD64 - unrelated question: is there a better hardware/os profile to build on?

tysonnorris, 2021-01-25:
Seems like using version 0.20.2 of libupnpp worked for compile, but then I got another error on nodejs modules:
```
Thank you for using serialport!
If you rely on this package, please consider supporting our open collective:
> https://opencollective.com/serialport/donate

+ eventemitter3@4.0.7
+ country-list@2.2.0
+ websocket@1.0.33
+ request@2.88.2
+ express@4.17.1
+ xml-js@1.6.11
+ dnssd2@1.0.0
+ underscore@1.12.0
+ wireless-tools@0.19.0
+ node-fetch@2.6.1
+ node-hid@2.1.1
+ serialport@9.0.6
+ jimp@0.16.1
+ pigpio@3.3.0
+ aplay@1.2.0
+ onoff@6.0.1
+ node-tunein-radio@0.3.4
+ node-cec@0.1.2
+ mpd-api@1.0.6
added 379 packages from 332 contributors in 35.481s
ERROR: architecture for "/usr/lib/node_modules/websocket/node_modules/utf-8-validate/prebuilds/linux-arm64/node.napi.node" is "AArch64", should be "ARM"
ERROR: architecture for "/usr/lib/node_modules/websocket/node_modules/bufferutil/prebuilds/linux-arm64/node.napi.node" is "AArch64", should be "ARM"
ERROR: architecture for "/usr/lib/node_modules/websocket/node_modules/utf-8-validate/prebuilds/linux-x64/node.napi.node" is "Advanced Micro Devices X86-64", should be "ARM"
ERROR: architecture for "/usr/lib/node_modules/websocket/node_modules/bufferutil/prebuilds/linux-x64/node.napi.node" is "Advanced Micro Devices X86-64", should be "ARM"
package/pkg-generic.mk:359: recipe for target '/home/ubuntu/buildroot-3/build/nodejs-12.18.4/.stamp_installed' failed
make[1]: *** [/home/ubuntu/buildroot-3/build/nodejs-12.18.4/.stamp_installed] Error 1
Makefile:84: recipe for target '_all' failed
make: *** [_all] Error 2
```

usul27, 2021-01-25:
That's just normal. Just start compile again.

MatteoEng, 2021-01-25:
Hi, how did you replace the 0.20.0 with the 0.20.2 of libupnpp? thanks

tysonnorris, 2021-01-25:
@MatteoEng I patched the patch file `buildroot/buildroot-2020.11.patch` and re-ran `get-buildroot`. 
I also ran into another similar missing version with upmpdcli, so updated that one to upmpdcli-1.5.8.tar.gz
I'll submit a PR if the compile completes, and it still needs updating.

```
$ git diff buildroot/buildroot-2020.11.patch
diff --git a/buildroot/buildroot-2020.11.patch b/buildroot/buildroot-2020.11.patch
index c6ed5a8..efb855a 100644
--- a/buildroot/buildroot-2020.11.patch
+++ b/buildroot/buildroot-2020.11.patch
@@ -209,7 +209,7 @@ diff -ru a/package/libupnp/libupnp.mk b/package/libupnp/libupnp.mk
  # Hash from: http://www.lesbonscomptes.com/upmpdcli/downloads/libupnpp-0.19.4.tar.gz.sha256
  sha256  29d1d07d44c8f79573e64e093ad8f0f883b11ef9eb3642a67aa703c1dde60576  libupnpp-0.19.4.tar.gz
 -sha256  f6e1016c1d3c9b0e065f2c72a86bb9538b2c6d8862ae010d97ba571f903622e2  libupnpp-b6ad3f1ce9fbc15364504f02ae2f3239a18af93b.tar.gz
-+sha256  5c75eb5d2336baec27be2342b4d53086cdd4f56a7756398a055028cb4a1eb00e  libupnpp-0.20.0.tar.gz
++sha256  48c2cc7b5a3eef55f4dddda88d8676ed11f7e88f1d6680c8458512067a0e0a11  libupnpp-0.20.2.tar.gz
  
  # Hash for license file:
  sha256  00a89b0d18aacd4114decf79122db87bf35bddaf2bc50e383c9c9f4c263390b2  COPYING
@@ -222,7 +222,7 @@ diff -ru a/package/libupnp/libupnp.mk b/package/libupnp/libupnp.mk
  
 -#LIBUPNPP_VERSION = 0.19.4
 -#LIBUPNPP_SITE = http://www.lesbonscomptes.com/upmpdcli/downloads
-+LIBUPNPP_VERSION = 0.20.0
++LIBUPNPP_VERSION = 0.20.2
 +LIBUPNPP_SITE = http://www.lesbonscomptes.com/upmpdcli/downloads
  LIBUPNPP_LICENSE = LGPL-2.1+
  LIBUPNPP_LICENSE_FILES = COPYING
@@ -243,7 +243,7 @@ diff -ru a/package/libupnp/libupnp.mk b/package/libupnp/libupnp.mk
  sha256  e3d90dedee6a942acc049324ca81b7f008d217d386a3e2590a2ca5670cc9df05  upmpdcli-1.4.14.tar.gz
 -sha256  0247c1d20069636eb72f3e11b4962cfc3acad0cce7c4815642c5896e9a236026  upmpdcli-master.tar.gz
 -sha256  ec570f345feeca4d8b2004150f3f2c46682170970387d8f8e6bbf4ea9af3faf5  upmpdcli-da758110f5f523fa2a313eec5e9a2878319ca60b.tar.gz
-+sha256  c971ed29cf00ebe70b6461f30ec4b9d58e0acaf52c6fd452876678bb6fe53009  upmpdcli-1.5.0.tar.gz
++sha256  802a2ab795841fbb9bfff5fd7951881022f5414ed893a1db704801b999e658a8  upmpdcli-1.5.8.tar.gz
  
  # Hash for license file:
  sha256  dc626520dcd53a22f727af3ee42c770e56c97a64fe3adb063799d8ab032fe551  COPYING
@@ -261,7 +261,7 @@ diff -ru a/package/libupnp/libupnp.mk b/package/libupnp/libupnp.mk
 -UPMPDCLI_SITE=https://framagit.org/medoc92/upmpdcli
 -UPMPDCLI_SITE_METHOD=git
 -UPMPDCLI_AUTORECONF = YES
-+UPMPDCLI_VERSION = 1.5.0
++UPMPDCLI_VERSION = 1.5.8
 +UPMPDCLI_SITE = http://www.lesbonscomptes.com/upmpdcli/downloads
  
  UPMPDCLI_LICENSE = LGPL-2.1+
```

MatteoEng, 2021-01-26:
yeah, thanks also yesterday I needed to patch the patch for libupnpp 0.20.2 and upmpdcli 1.5.8, I hope now will complete the compilation

hifiberry, 2021-02-02:
The PR has been merged into the development branch

PR #230 (2021-02-01): update patch versions of libupnpp and upmpdcli
Fixes #229 
I also had trouble with a missing `images` directory during build, so updated the `create-image` script to create that
closed 2021-02-02 by hifiberry