hifiberry/hifiberry-os #231: Equaliser has no effect on Snapcast

depressivum, 2021-01-28:
**Describe the bug**
While the Euqualizer works for radio, it does not for Snapcast.

**HiFiBerryOS version**
Systemsoftware 20201213

**HiFiBerry sound card**
Amp2

**To Reproduce**
Steps to reproduce the behavior:
1. Select Source "Snapcast"
2. Click on "Sound / Equaliser"
3. Change default settings
4. The Sound remains the same

**Expected behavior**
The Sound is changing like other sources (e.g. Radio)

**Screenshots**
-

**Browser (if applicable)**
 - OS: Windows 10 / Android 9.0
 - Browser: Firefox 
 - Version 84.0.2
 - 
**Additional context**
-

hifiberry, 2021-02-01:
That's possible. As the equalizer creates lots of issues also with other players, we'll disable it in the next release. If somebody wants to contribute and dig into this, we're open to contributions.

closed 2021-02-01 by hifiberry

depressivum, 2021-02-02:
Okay, this would be sad as i like the equaliser very much. The speakers i use aren´t useable without it so i would appreciate this feature to stay, at least in experimental state.

tysonnorris, 2021-02-02:
I don't use snapcast myself, but would be happy to help with disabling the equaliser in a way that allows it to remain operational for people willing to take the risk, while disabling it by default. 
I guess this would primarily be making the ui only expose the equaliser "enable" button if some config flag is in place? (enabling the enable button)

hifiberry, 2021-02-03:
Eq can still be enabled from command line.