hifiberry/hifiberry-os #233: Podcasts cannot be played via Spotify Connect

timlapluie, 2021-02-02:
**Describe the bug**
When playing a podcast in Spotify, it is not possible to connect to HiFiBerry via Spotify Connect. HiFiBerry is greyed out in the available Spotify Connect devices with the message `Can't play this right now`. 

**HiFiBerryOS version**
20201213

**HiFiBerry sound card**
Any HiFiBerry card

**To Reproduce**
Steps to reproduce the behavior:
1. Open Spotify (Premium)
2. Play a Podcast
3. Try to connect to HiFiBerry via Spotify Connect

**Expected behavior**
Podcasts are played via HifiBerry, just like all other songs

**Screenshots**
(`HiFiStereo` is a HiFiBerry with Spotify Connect)
[image]

**Additional context**
Looking into the fork of librespot that is used by [`vollibrespot`](https://github.com/ashthespy/vollibrespot) (https://github.com/ashthespy/librespot), it is far behind the latest version of its origin (https://github.com/librespot-org/librespot) that added Podcast support in October 2019 (https://github.com/librespot-org/librespot/pull/381).

Maybe it is possible somehow to bump `librespot` in order to allow podcasts to work on HiFiBerry OS.

hifiberry, 2021-02-03:
Upgrading this would be major work and we probably won't look into this soon. But we're open for contributions.

timlapluie, 2021-02-04:
Haven't looked into it further, but it seems to be an intermittent issue. 
Podcasts seem to work as long as you start playing any song (not Podcast), stream it to HiFiBerry via Spotify Connect and then switch to a Podcast. At least there is a workaround!

closed 2021-07-22 by hifiberry