hifiberry/hifiberry-os #236: cannot connect to spotify with latest development branch

tysonnorris, 2021-02-08:
DO NOT remove the blocks below, but fill these with the requested data. Incomplete bug reports will be ignored!
You should remove this comment before posting the bug report.

**Describe the bug**
A clear and concise description of what the bug is.

When I initiate spotify connect, the connection fails. Logs from journalctl are 
```
Feb 07 22:06:21 hifiberry vollibrespot[777]: [Vollibrespot] : Connecting to AP "guc3-accesspoint-e-d4ln.ap.spotify.com:443"
Feb 07 22:06:22 hifiberry vollibrespot[777]: [Vollibrespot] : Authenticated as "noizenheimer" !
Feb 07 22:06:22 hifiberry vollibrespot[777]: [Vollibrespot] : Setting up new mixer: card:hw:0 mixer:Digital index:0
Feb 07 22:06:22 hifiberry vollibrespot[777]: [Vollibrespot] : Alsa Mixer info min: 0 (MilliBel(-9999999)[dB]) -- max: 207 (MilliBel(0)[dB]) HW: true
Feb 07 22:06:22 hifiberry vollibrespot[777]: [Vollibrespot] : Using alsa sink
Feb 07 22:06:22 hifiberry vollibrespot[777]: [Vollibrespot] : Metadata pipe established
Feb 07 22:06:22 hifiberry vollibrespot[777]: [Vollibrespot] : Event: Volume { volume_to_mixer: 32767 }
Feb 07 22:06:22 hifiberry python3[608]: ERROR:root:Don't know what to do with vollibrespot v0.2.2
Feb 07 22:06:22 hifiberry vollibrespot[777]: [Vollibrespot] : Country: "US"
Feb 07 22:06:22 hifiberry vollibrespot[777]: [Vollibrespot] : Fetching autoplay context uri
Feb 07 22:06:22 hifiberry vollibrespot[777]: [Vollibrespot] : Event: SessionActive { became_active_at: 1612764382789 }
Feb 07 22:06:22 hifiberry vollibrespot[777]: [Vollibrespot] : SessionActive!
Feb 07 22:06:22 hifiberry vollibrespot[777]: [Vollibrespot] : Event: PlaybackStarted { track_id: SpotifyId { id: 227532073366269524790086497955504360008, audio_type: Track } }
Feb 07 22:06:22 hifiberry vollibrespot[777]: [Vollibrespot] : Event: PlaybackLoading { track_id: SpotifyId { id: 227532073366269524790086497955504360008, audio_type: Track } }
Feb 07 22:06:22 hifiberry vollibrespot[777]: [Vollibrespot] : Autoplay uri resolved to <"spotify:station:playlist:37i9dQZF1E39q8jD9iE3KC">
Feb 07 22:06:22 hifiberry vollibrespot[777]: [Vollibrespot] : Resolving uri "spotify:station:playlist:37i9dQZF1E39q8jD9iE3KC"
Feb 07 22:06:22 hifiberry vollibrespot[777]: [Vollibrespot] : error 400 for uri hm://keymaster/token/authenticated?client_id=&scope=streaming,user-read-playback-state,user-modify-playback-state,user-read-currently-playing,user-read-private
Feb 07 22:06:22 hifiberry vollibrespot[777]: [Vollibrespot] : Error: MercuryError
Feb 07 22:06:22 hifiberry vollibrespot[777]: thread 'main' panicked at 'cannot poll Map twice', /home/ubuntu/buildroot-3/host/share/cargo/registry/src/github.com-1ecc6299db9ec823/futures-0.1.29/src/future/map.rs:35:15
Feb 07 22:06:22 hifiberry vollibrespot[777]: stack backtrace:
Feb 07 22:06:22 hifiberry vollibrespot[777]:    0:   0xa284dc - <unknown>
Feb 07 22:06:22 hifiberry vollibrespot[777]:    1:   0xa50698 - <unknown>
Feb 07 22:06:22 hifiberry vollibrespot[777]:    2:   0xa2153c - <unknown>
Feb 07 22:06:22 hifiberry vollibrespot[777]:    3:   0xa2abcc - <unknown>
Feb 07 22:06:22 hifiberry vollibrespot[777]:    4:   0xa2a880 - <unknown>
Feb 07 22:06:22 hifiberry vollibrespot[777]:    5:   0xa2b284 - <unknown>
Feb 07 22:06:22 hifiberry vollibrespot[777]:    6:   0xa2aea0 - <unknown>
Feb 07 22:06:22 hifiberry vollibrespot[777]:    7:   0xa4dd70 - <unknown>
Feb 07 22:06:22 hifiberry vollibrespot[777]:    8:   0xa4db4c - <unknown>
Feb 07 22:06:22 hifiberry vollibrespot[777]:    9:   0x707ca4 - <unknown>
Feb 07 22:06:22 hifiberry vollibrespot[777]:   10:   0x5a7d88 - <unknown>
Feb 07 22:06:22 hifiberry vollibrespot[777]:   11:   0x52cef0 - <unknown>
Feb 07 22:06:22 hifiberry vollibrespot[777]:   12:   0x4f3d24 - <unknown>
Feb 07 22:06:22 hifiberry vollibrespot[777]:   13:   0x508ac0 - <unknown>
Feb 07 22:06:22 hifiberry vollibrespot[777]:   14:   0x4e693c - <unknown>
Feb 07 22:06:22 hifiberry vollibrespot[777]:   15:   0x5097a8 - <unknown>
Feb 07 22:06:22 hifiberry vollibrespot[777]:   16:   0x51c940 - <unknown>
Feb 07 22:06:22 hifiberry vollibrespot[777]:   17:   0x4e4b6c - <unknown>
Feb 07 22:06:22 hifiberry vollibrespot[777]:   18:   0x52e578 - <unknown>
Feb 07 22:06:22 hifiberry vollibrespot[777]:   19:   0x51d870 - <unknown>
Feb 07 22:06:22 hifiberry vollibrespot[777]:   20:   0xa2b648 - <unknown>
Feb 07 22:06:22 hifiberry vollibrespot[777]:   21:   0x52f030 - <unknown>
Feb 07 22:06:22 hifiberry vollibrespot[777]:   22: 0x76d396c4 - __libc_start_main
```

**HiFiBerryOS version**
You find this on the "About" screen. It looks like Software 20YYMMDD
20210208

**HiFiBerry sound card**
Post the exact type, e.g. DAC+ Pro instead of "DAC+".
DAC+/Amp2

**To Reproduce**
Steps to reproduce the behavior:
1. startup system
2. attempt to connect from mobile via spotify connect
3. connection fails

**Expected behavior**
A clear and concise description of what you expected to happen.
Connection should not fail

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Browser (if applicable)**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

Another system running previous release version (failure is on development version), I see a token result in logs:
```
Feb 07 21:45:34 hifiberry-pi3 vollibrespot[572]: [Vollibrespot] : Autoplay uri resolved to <"spotify:station:playlist:37i9dQZF1E39q8jD9iE3KC">
Feb 07 21:45:34 hifiberry-pi3 vollibrespot[572]: [Vollibrespot] : Resolving uri "spotify:station:playlist:37i9dQZF1E39q8jD9iE3KC"
Feb 07 21:45:34 hifiberry-pi3 vollibrespot[572]: [Vollibrespot] : Event: GotToken { token: Token { access_token: "BQBEfvphlw1SDGZ0PZQZaqzKEGY67dnbPP_jVB6YQNQPbfMQPnpW_G1zq_ClpF_apDnzLE5wGtfgJT4c_O3u6PbX83UStA2rkENvxg_rbEw6vaRVbFumV88MTv7ikevxLmmtc4Vab4IVJAVpRes4pIO_Wkub_rK0Z56HPOtS", expires_in: 3600, token_type: "Bearer", scope: ["streaming", "user-read-playback-state", "user-modify-playback-state", "user-read-currently-playing", "user-read-private"] } }
Feb 07 21:45:34 hifiberry-pi3 vollibrespot[572]: [Vollibrespot] : Loading <Give It Back> with Spotify URI <spotify:track:5d0i2Pu2DlNNsFbPJYmpKg>
```
similar lines during the development version failure fail with a 400 at the spotify token api
```
Feb 08 05:39:26 hifiberry vollibrespot[2788]: [Vollibrespot] : Autoplay uri resolved to <"spotify:station:playlist:37i9dQZF1E39q8jD9iE3KC">
Feb 08 05:39:26 hifiberry vollibrespot[2788]: [Vollibrespot] : Resolving uri "spotify:station:playlist:37i9dQZF1E39q8jD9iE3KC"
Feb 08 05:39:26 hifiberry vollibrespot[2788]: [Vollibrespot] : error 400 for uri hm://keymaster/token/authenticated?client_id=&scope=streaming,user-read-playback-state,user-modify-playback-state,user-read-currently-playing,user-read-private
Feb 08 05:39:26 hifiberry vollibrespot[2788]: [Vollibrespot] : Error: MercuryError
```

This seems like a vollibrespot issue, but I cannot find any indication it (version/configs/etc) has changed since previous release.

usul27, 2021-02-08:
I don't see this here. We just created a new build from the development tree that works fine here. You might try this one:

/opt/hifiberry/bin/update --experimental

tysonnorris, 2021-02-08:
Running `/opt/hifiberry/bin/update --experimental` will not update because it says the current version is newer.

I tried running an image built from master 20210125, and it had the same problem. 

I will also try running previous release 20201213 and then attempt to apply the development update using `/opt/hifiberry/bin/update --experimental`.

So far seems like something I'm doing wrong to get my built images, since I only see minor changes in master and development, and the build seems to go OK; and the previous release image seems to work OK.

tysonnorris, 2021-02-09:
So I ran 20201213, then applied `/opt/hifiberry/bin/update --experimental` - the update went fine, and spotify was still fine. Verified new version shows in the ui.

Next I did a fresh build from development branch (as of Feb 8 2021), and same problem, spotify cannot connect. 

Now I'm wondering if any of my built images have spotify working. I will go back to master branch and the v20201213 tag, to see if either of those works when build from source. 

I can't figure what might be breaking this in my build though, I'm just doing:
```
./get-buildroot
./build-config 3
./compile 3
#crashes on first compile of node modules; retry is successful
./compile 3
./create-image 3
```
and the image is created without any issue, other than the compile always failing on node modules on the first run.

tysonnorris, 2021-02-09:
ah I think this is the problem https://github.com/hifiberry/hifiberry-os/blob/7f5fae5fd846d28706cbe565c271ebcfebaa5c5f/compile#L6

I guess I should read carefully the build output: 
```
Secrets file not found, this isn't an official HiFiBerry build
some components might not work as expected
```

I'll create a PR for build instructions to include the secrets setup to help anyone else that might run into this.

PR #237 (2021-02-10): mention secrets in building.md
I didn't notice that I needed a secrets file, so thought this might help others.
Fixes #236
closed 2021-02-10 by hifiberry