hifiberry/hifiberry-os #243: low volume when using snapcast

fyksen, 2021-02-23:
**Describe the bug**
Low volume when playing from snapcast.

**HiFiBerryOS version**
DAC+/Amp2
System software 20210203

**HiFiBerry sound card**
AMP 100 and amp+ on raspberry pi 4b

**To Reproduce**
Enable snapcast source. When playing from other sources i mostly is 20% of max audio. When using snapcast it is 70-80%. Breaks changing sources.

**Expected behavior**
Volume being roughly the same.

**Additional context**
Add any other context about the problem here.

Bertware, 2021-04-05:
The alsaloop source has a similar issue, although not as severe. Alsaloop requires about 20% more volume when receiving analog input at maximum volume (just before the ADC starts clipping at -9db), but this results in spotify playing way too loud the next time. Some sort of "volume per source", or volume adjustment per source, could resolve this for all sources at once.

hifiberry, 2021-04-06:
@Bertware While this might be the right way, the implementation would be very complex. One of the main issues is that the "%" values for volume don't really make sense, but unfortunately they are used by most programs :(
The right way would be to use decibels, but not all software and device drivers support this. As long somebody comes up with same good idea and implementation this probably will stay like this.

closed 2024-01-12 by hifiberry