hifiberry/hifiberry-os #244: localbrowser/cog generates invalid JSON messages

tysonnorris, 2021-02-28:
I've been attempting to modify an extension on beocreate2, and notice that the localbrowser (I have a connected display) generates invalid websocket messages regularly but not consistently. I don't see invalid messages from my laptop (macos + chrome) browser, which is why I'm logging the issue here instead of the create repo.

**Describe the bug**
A clear and concise description of what the bug is.

I added a line to dump messages before JSON.parse and when I restart cog service I see things like below; it is not the same invalid characters in the JSON, and not always there, but very often. Loading the UI on my laptop does not appear to generate the invalid messages.
```
Feb 27 17:14:51 hifiberry beocreate2[634]: message:{"type":"utf8","utf8Data":"{\"target\":\"product-information\",\"header\":\"getBasicProductInformation\"}0136A"}
Feb 27 17:14:51 hifiberry beocreate2[634]: Error in processing received data: SyntaxError: Unexpected number in JSON at position 70
Feb 27 17:14:51 hifiberry beocreate2[634]:     at JSON.parse (<anonymous>)
Feb 27 17:14:51 hifiberry beocreate2[634]:     at WebSocketConnection.<anonymous> (/opt/beocreate/beocreate_essentials/communication.js:115:24)
Feb 27 17:14:51 hifiberry beocreate2[634]:     at WebSocketConnection.emit (events.js:315:20)
Feb 27 17:14:51 hifiberry beocreate2[634]:     at WebSocketConnection.processFrame (/usr/lib/node_modules/websocket/lib/WebSocketConnection.js:554:26)
Feb 27 17:14:51 hifiberry beocreate2[634]:     at /usr/lib/node_modules/websocket/lib/WebSocketConnection.js:323:40
Feb 27 17:14:51 hifiberry beocreate2[634]:     at processTicksAndRejections (internal/process/task_queues.js:79:11)
Feb 27 17:14:51 hifiberry beocreate2[634]: message:{"type":"utf8","utf8Data":"{\"target\":\"sound\",\"header\":\"getVolume\"}tP"}
Feb 27 17:14:51 hifiberry beocreate2[634]: Error in processing received data: SyntaxError: Unexpected token t in JSON at position 39
Feb 27 17:14:51 hifiberry beocreate2[634]:     at JSON.parse (<anonymous>)
Feb 27 17:14:51 hifiberry beocreate2[634]:     at WebSocketConnection.<anonymous> (/opt/beocreate/beocreate_essentials/communication.js:115:24)
Feb 27 17:14:51 hifiberry beocreate2[634]:     at WebSocketConnection.emit (events.js:315:20)
Feb 27 17:14:51 hifiberry beocreate2[634]:     at WebSocketConnection.processFrame (/usr/lib/node_modules/websocket/lib/WebSocketConnection.js:554:26)
Feb 27 17:14:51 hifiberry beocreate2[634]:     at /usr/lib/node_modules/websocket/lib/WebSocketConnection.js:323:40
Feb 27 17:14:51 hifiberry beocreate2[634]:     at processTicksAndRejections (internal/process/task_queues.js:79:11)
```

**HiFiBerryOS version**
You find this on the "About" screen. It looks like Software 20YYMMDD

20210209

**HiFiBerry sound card**
Post the exact type, e.g. DAC+ Pro instead of "DAC+".
Amp2

**To Reproduce**
Steps to reproduce the behavior:
1. restart cog service
2. check beocreate2 logs
3. Notice `JSON.parse` errors
4. add a line to dump the message to console to see the message has extra chars

**Expected behavior**
A clear and concise description of what you expected to happen.

Expected no invalid messages.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Browser (if applicable)**
 - OS: hifiberry-os
 - Browser localbrowser/cog
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

tysonnorris, 2021-02-28:
I get different invalid characters in the same message source at different times:
```
# journalctl -u beocreate2 --no-pager | grep -C 1 JSON | grep -C 1 getSources
--
Feb 27 17:05:36 hifiberry beocreate2[634]: message:{"type":"utf8","utf8Data":"{\"target\":\"sources\",\"header\":\"getSources\"}na!"}
Feb 27 17:05:36 hifiberry beocreate2[634]: Error in processing received data: SyntaxError: Unexpected token n in JSON at position 42
--
--
Feb 27 17:12:11 hifiberry beocreate2[634]: message:{"type":"utf8","utf8Data":"{\"target\":\"sources\",\"header\":\"getSources\"}ra\u0011"}
Feb 27 17:12:11 hifiberry beocreate2[634]: Error in processing received data: SyntaxError: Unexpected token r in JSON at position 42
--
--
Feb 27 17:14:25 hifiberry beocreate2[634]: message:{"type":"utf8","utf8Data":"{\"target\":\"sources\",\"header\":\"getSources\"}1B5"}
Feb 27 17:14:25 hifiberry beocreate2[634]: Error in processing received data: SyntaxError: Unexpected number in JSON at position 42
--
--
Feb 27 17:25:05 hifiberry beocreate2[1608]: message:{"type":"utf8","utf8Data":"{\"target\":\"sources\",\"header\":\"getSources\"}ep\u0011"}
Feb 27 17:25:05 hifiberry beocreate2[1608]: Error in processing received data: SyntaxError: Unexpected token e in JSON at position 42
```

akelge, 2021-03-08:
I think I have found another related issue.
If I try to change volume using a touchscreen it fails and in cog logs I can see

```Mar 08 16:44:50 the-musical-box WPENetworkProce[871]: soup_websocket_connection_send_text: assertion 'g_utf8_validate (text, length, NULL)' failed```

hifiberry, 2021-03-17:
Yes, this is a bug, but we don't know yet why it happens and how to deal with it. If somebody ist experienced with websocket communication in Javascript, we're interested in some help. For now it looks like a bug in the Javascript implementation of the browser, but it might be a bug in the UI code itself that we don't see yet.

hifiberry, 2021-03-19:
We found a workaround for this. This will be included in the next release.

tysonnorris, 2021-03-29:
Thanks! Is there a PR that you can point to that I can experiment with to verify it is fixed in my setup?

hifiberry, 2021-11-08:
Fixed

closed 2021-11-08 by hifiberry