hifiberry/hifiberry-os #246: Set Codec order in squeezelite

gaetanars, 2021-03-08:
**Is your feature request related to a problem? Please describe.**
In my home I force some of my squeezelite to run in PCM only (my LMS convert FLAC or MP3 to PCM). For this I run Squeezelite with `-c pcm,flac,mp3`

**Describe the solution you'd like**
Is it possible to add this parameters on HifiberryOS interface?

**Describe alternatives you've considered**

**Additional context**

usul27, 2021-03-08:
If we add this to the UI, it would end up like some other distributions that have tons of configuration options that most users don't understand. We won't add this to the GUI, but you can edit the configuration file directly.

gaetanars, 2021-03-08:
Ok I understand this point of view. This is also this settings that can be add in an « advanced settings » section.

closed 2021-03-08 by gaetanars