hifiberry/hifiberry-os #249: Cannot ping external hostnames

kristojorg, 2021-03-11:
**Describe the bug**
My hifiberry is not able to ping external hostnames, which is causing Spotify connect to fail (I believe):
```
# ping google.com
ping: bad address 'google.com'
```

It is able to ping an external IP address:
```
# ping 8.8.8.8
PING 8.8.8.8 (8.8.8.8): 56 data bytes
64 bytes from 8.8.8.8: seq=0 ttl=119 time=15.417 ms
```

**Some notes**:
- I have freshly re-flashed the SD card and completed setup with ethernet attached.
- My computer attached to the same network can ping external hostnames.
- I initially thought this was due to the date being wrong, which happened the first time I set up the pi over wifi. Since setting it up over ethernet, the date is correct and properly sync'd with ntp, but the problem persists. 

**HiFiBerryOS version**
20210203

**HiFiBerry sound card**
Amp 2

**To Reproduce**
Steps to reproduce the behavior:
1. Install latest version Hifiberry OS (may also happen on old versions, idk)
2. SSH into the pi `ssh root@hifiberry.local`
3. Attempt to ping an external hostname `ping google.com`
4. See error

**Expected behavior**
Should be able to ping external hostname.

**Additional Context**

I discovered this issue because Spotify connect was not working, and logging gave me this:

```
Mar 11 09:52:51 hifiberry systemd[1]: Started Vollibrespot.
Mar 11 09:55:24 hifiberry vollibrespot[917]: [Vollibrespot] : couldn't parse packet from V4(192.168.8.18:5353): query type 65 is invalid
Mar 11 09:55:24 hifiberry vollibrespot[917]: [Vollibrespot] : couldn't parse packet from V6([fe80::83f:554c:ad9b:eb85]:5353): query type 65 is invalid
Mar 11 09:55:25 hifiberry vollibrespot[917]: [Vollibrespot] : couldn't parse packet from V4(192.168.8.18:5353): query type 65 is invalid
Mar 11 09:55:25 hifiberry vollibrespot[917]: [Vollibrespot] : couldn't parse packet from V6([fe80::83f:554c:ad9b:eb85]:5353): query type 65 is invalid
Mar 11 09:55:28 hifiberry vollibrespot[917]: [Vollibrespot] : couldn't parse packet from V4(192.168.8.18:5353): query type 65 is invalid
Mar 11 09:55:28 hifiberry vollibrespot[917]: [Vollibrespot] : couldn't parse packet from V6([fe80::83f:554c:ad9b:eb85]:5353): query type 65 is invalid
Mar 11 09:55:37 hifiberry vollibrespot[917]: [Vollibrespot] : couldn't parse packet from V4(192.168.8.18:5353): query type 65 is invalid
Mar 11 09:55:37 hifiberry vollibrespot[917]: [Vollibrespot] : couldn't parse packet from V6([fe80::83f:554c:ad9b:eb85]:5353): query type 65 is invalid
Mar 11 09:56:04 hifiberry vollibrespot[917]: [Vollibrespot] : couldn't parse packet from V4(192.168.8.18:5353): query type 65 is invalid
Mar 11 09:56:04 hifiberry vollibrespot[917]: [Vollibrespot] : couldn't parse packet from V6([fe80::83f:554c:ad9b:eb85]:5353): query type 65 is invalid
Mar 11 09:57:51 hifiberry vollibrespot[917]: [Vollibrespot] : couldn't parse packet from V4(192.168.8.18:5353): query type 65 is invalid
Mar 11 09:57:51 hifiberry vollibrespot[917]: [Vollibrespot] : couldn't parse packet from V6([fe80::83f:554c:ad9b:eb85]:5353): query type 65 is invalid
Mar 11 09:57:52 hifiberry vollibrespot[917]: [Vollibrespot] : couldn't parse packet from V4(192.168.8.18:5353): query type 65 is invalid
Mar 11 09:57:52 hifiberry vollibrespot[917]: [Vollibrespot] : couldn't parse packet from V6([fe80::83f:554c:ad9b:eb85]:5353): query type 65 is invalid
Mar 11 09:57:55 hifiberry vollibrespot[917]: [Vollibrespot] : couldn't parse packet from V4(192.168.8.18:5353): query type 65 is invalid
Mar 11 09:57:55 hifiberry vollibrespot[917]: [Vollibrespot] : couldn't parse packet from V6([fe80::83f:554c:ad9b:eb85]:5353): query type 65 is invalid
Mar 11 09:58:46 hifiberry vollibrespot[917]: [Vollibrespot] : couldn't parse packet from V4(192.168.8.18:5353): query type 65 is invalid
Mar 11 09:58:46 hifiberry vollibrespot[917]: [Vollibrespot] : couldn't parse packet from V6([fe80::83f:554c:ad9b:eb85]:5353): query type 65 is invalid
Mar 11 09:58:47 hifiberry vollibrespot[917]: [Vollibrespot] : couldn't parse packet from V4(192.168.8.18:5353): query type 65 is invalid
Mar 11 09:58:47 hifiberry vollibrespot[917]: [Vollibrespot] : couldn't parse packet from V6([fe80::83f:554c:ad9b:eb85]:5353): query type 65 is invalid
Mar 11 09:58:50 hifiberry vollibrespot[917]: [Vollibrespot] : couldn't parse packet from V4(192.168.8.18:5353): query type 65 is invalid
Mar 11 09:58:50 hifiberry vollibrespot[917]: [Vollibrespot] : couldn't parse packet from V6([fe80::83f:554c:ad9b:eb85]:5353): query type 65 is invalid
Mar 11 09:59:13 hifiberry vollibrespot[917]: [Vollibrespot] : couldn't parse packet from V4(192.168.8.18:5353): query type 65 is invalid
Mar 11 09:59:13 hifiberry vollibrespot[917]: [Vollibrespot] : couldn't parse packet from V6([fe80::83f:554c:ad9b:eb85]:5353): query type 65 is invalid
Mar 11 09:59:14 hifiberry vollibrespot[917]: [Vollibrespot] : couldn't parse packet from V4(192.168.8.18:5353): query type 65 is invalid
Mar 11 09:59:14 hifiberry vollibrespot[917]: [Vollibrespot] : couldn't parse packet from V6([fe80::83f:554c:ad9b:eb85]:5353): query type 65 is invalid
Mar 11 09:59:17 hifiberry vollibrespot[917]: [Vollibrespot] : couldn't parse packet from V4(192.168.8.18:5353): query type 65 is invalid
Mar 11 09:59:17 hifiberry vollibrespot[917]: [Vollibrespot] : couldn't parse packet from V6([fe80::83f:554c:ad9b:eb85]:5353): query type 65 is invalid
Mar 11 10:06:24 hifiberry vollibrespot[917]: [Vollibrespot] : Failed to resolve Access Point: HTTP error
Mar 11 10:06:24 hifiberry vollibrespot[917]: [Vollibrespot] : Using fallback "ap.spotify.com:443"
Mar 11 10:06:24 hifiberry vollibrespot[917]: [Vollibrespot] : Connecting to AP "ap.spotify.com:443"
Mar 11 10:06:24 hifiberry vollibrespot[917]: thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: Custom { kind: Other, error: "failed to lo
Mar 11 10:06:24 hifiberry vollibrespot[917]: stack backtrace:
```

I am still able to use airplay and bluetooth, and sound output works, but spotify connect fails.

kristojorg, 2021-03-11:
I also see some logs in `journalctl` about failed DNSSEC earlier on:
```
Mar 11 09:52:06 hifiberry systemd-resolved[181]: DNSSEC validation failed for question pool.ntp.org IN DS: failed-auxiliary
Mar 11 09:52:06 hifiberry systemd-resolved[181]: DNSSEC validation failed for question 0.pool.ntp.org IN AAAA: failed-auxiliary
Mar 11 09:52:06 hifiberry systemd-resolved[181]: DNSSEC validation failed for question google.com IN SOA: failed-auxiliary
```
Errors like these are pretty heavily spread throughout the logs.

kristojorg, 2021-03-11:
**Update with fix:**

I was able to fix this issue, it turns out the date/time was indeed the problem. Here are the steps:

1. You must setup hifiberry with ethernet. If you do not, the hifiberry will fail to access the NTP server used to set date/time because DNSSEC will fail due to the date being wrong. And such you have a loop. I tried fixing this by manually setting date/time but that didn't work. So use ethernet when first setting up, this somehow allows it to properly access the NTP server and change the date/time.
2. Restart pi. The time (in UTC) from step 1 was still wrong by a few minutes for some reason. 
3. The timezone is still wrong. Set your timezone: `timedatectl set-timezone "Europe/Paris"`. 
4. Restart pi.
5. You can now ping an external hostname because DNSSEC won't fail since your time is in the appropriate range. You can now use Spotify connect! Enjoy : )

kristojorg, 2021-03-11:
**Correction:**

Well, that fixed it while it was connected to ethernet. Now that I have unplugged the pi, disconnected ethernet and moved it to its final location (no ethernet available), we are back at step 1. The time is wrong (due to the time it was powered off for), and it cannot ping external hostnames again.

I think the solution to this might be to disable DNSSEC so that the pi can reach the NTP server and set its own time, but I'm not sure and I don't quite know how to accomplish that.

hifiberry, 2021-03-11:
Yeah, DNSSec is a chicken-and-egg issue on the Pi: Without the correct time, it doesn't work, but we can't set the correct time without DNSSec ;-)

However, DNSSec should be disabled temporary to allow initial time sync by this script:

https://github.com/hifiberry/hifiberry-os/blob/4fff8527510b66c2dfb1d0beb521f1f3e4da8c63/buildroot/package/hifiberry-systemd/fix-ntp.sh

kristojorg, 2021-03-11:
Thanks for filling me in on that, can you tell me how to run the script? I can't seem to find where it is on the disk

hifiberry, 2021-03-11:
This should be in /opt/hifiberry/bin and should be started automatically

https://github.com/hifiberry/hifiberry-os/blob/4fff8527510b66c2dfb1d0beb521f1f3e4da8c63/buildroot/package/hifiberry-systemd/hifiberry-systemd.mk

Not sure, why it doesn't on your system

kristojorg, 2021-03-11:
Ok I found it and ran it, but the date is still incorrect. Is there something additional I need to do to actually trigger the date sync?

hifiberry, 2021-03-11:
restart ntpd

kristojorg, 2021-03-11:
I ran this command, but still getting the error: 
```
# systemctl restart ntpd
# date
Thu Mar 11 13:55:02 CET 2021
# date
Thu Mar 11 13:55:05 CET 2021
# ping google.com
ping: bad address 'google.com'
#
```

Additionally, if I try to update the ntp server manually:
```
# ntpdate -u server 0.fr.pool.ntp.org
Error resolving server: Name or service not known (-2)
11 Mar 13:53:31 ntpdate[3405]: Can't find host server: Name or service not known (-2)
Error resolving 0.fr.pool.ntp.org: Name or service not known (-2)
11 Mar 13:53:32 ntpdate[3405]: Can't find host 0.fr.pool.ntp.org: Name or service not known (-2)
11 Mar 13:53:32 ntpdate[3405]: no servers can be used, exiting
```

It seems I'm still in the chicken and egg situation... any thoughts?

kristojorg, 2021-03-11:
I have connected the pi to my phone's hotspot and ping succeeded on external hostnames. There must be something wrong with my router / LAN. I will try to dig in to that.

kristojorg, 2021-03-11:
What's strange is that my computer is on the same network and can ping external hostnames totally fine...

fdiamant, 2021-03-13:
I have the same problem with the difference that I can ping external hostnames. I have noticed that if I do a clean install, Spotify connect works the first time. After that, I get the "couldn't parse packet error". Sometimes, which seems random to me, it does connect.

`PING 8.8.8.8 (8.8.8.8): 56 data bytes
64 bytes from 8.8.8.8: seq=0 ttl=117 time=55.830 ms
64 bytes from 8.8.8.8: seq=1 ttl=117 time=55.784 ms
64 bytes from 8.8.8.8: seq=2 ttl=117 time=55.516 ms

--- 8.8.8.8 ping statistics ---
4 packets transmitted, 3 packets received, 25% packet loss
round-trip min/avg/max = 55.516/55.710/55.830 ms`

`PING google.com (142.250.186.110): 56 data bytes
64 bytes from 142.250.186.110: seq=0 ttl=116 time=55.168 ms
64 bytes from 142.250.186.110: seq=1 ttl=116 time=54.917 ms
64 bytes from 142.250.186.110: seq=2 ttl=116 time=55.179 ms

--- google.com ping statistics ---
4 packets transmitted, 3 packets received, 25% packet loss
round-trip min/avg/max = 54.917/55.088/55.179 ms`

Time is right

`               Local time: Sat 2021-03-13 13:54:06 EET
           Universal time: Sat 2021-03-13 11:54:06 UTC
                 RTC time: n/a                        
                Time zone: Europe/Athens (EET, +0200) 
System clock synchronized: yes                        
              NTP service: active                     
          RTC in local TZ: no      `

`● spotify.service - Vollibrespot
     Loaded: loaded (/usr/lib/systemd/system/spotify.service; enabled; vendor preset: enabled)
     Active: active (running) since Sat 2021-03-13 13:52:23 EET; 58s ago
    Process: 1083 ExecStartPre=/opt/hifiberry/bin/bootmsg Starting Vollibrespot (code=exited, status=0/SUCCESS)
    Process: 1087 ExecStartPre=/opt/hifiberry/bin/store-volume /tmp/spotifyvol (code=exited, status=0/SUCCESS)
    Process: 1093 ExecStartPost=/usr/bin/sleep 1 (code=exited, status=0/SUCCESS)
    Process: 1094 ExecStartPost=/opt/hifiberry/bin/restore-volume /tmp/spotifyvol (code=exited, status=0/SUCCESS)
    Process: 1108 ExecStartPost=/usr/bin/rm /tmp/spotifyvol (code=exited, status=0/SUCCESS)
   Main PID: 1092 (vollibrespot)
      Tasks: 1 (limit: 3804)
     CGroup: /system.slice/spotify.service
             └─1092 /usr/bin/vollibrespot -c /etc/vollibrespot.conf

Mar 13 13:52:26 solaria vollibrespot[1092]: [Vollibrespot] : couldn't parse packet from V4(192.168.1.104:5353): query type 65 is invalid
Mar 13 13:52:35 solaria vollibrespot[1092]: [Vollibrespot] : couldn't parse packet from V4(192.168.1.104:5353): query type 65 is invalid
Mar 13 13:52:36 solaria vollibrespot[1092]: [Vollibrespot] : couldn't parse packet from V4(192.168.1.104:5353): query type 65 is invalid
Mar 13 13:52:39 solaria vollibrespot[1092]: [Vollibrespot] : couldn't parse packet from V4(192.168.1.104:5353): query type 65 is invalid
Mar 13 13:52:48 solaria vollibrespot[1092]: [Vollibrespot] : couldn't parse packet from V4(192.168.1.104:5353): query type 65 is invalid
Mar 13 13:52:59 solaria vollibrespot[1092]: [Vollibrespot] : couldn't parse packet from V4(192.168.1.104:5353): query type 65 is invalid
Mar 13 13:53:00 solaria vollibrespot[1092]: [Vollibrespot] : couldn't parse packet from V4(192.168.1.104:5353): query type 65 is invalid`

hifiberry, 2021-03-15:
@fdiamant : The "could not parse packet errors" are just normal. This has nothing to do at all with this bug report.

kristojorg, 2021-03-25:
After not being able to log in to my crappy router my ISP gave me, I bought a new one and now this works. Seems like it must have been a LAN issue. Thanks for the help here

closed 2021-03-25 by kristojorg