hifiberry/hifiberry-os #251: Occasionally can't control volume with phone

kristojorg, 2021-03-25:
**Describe the bug**
Most of the time, when playing via Spotify Connect, when I change my phone volume, the pi volume changes. Sometimes, however, even when I have spotify open on my screen and I change the volume nothing happens. Usually it will show the little spotify connect volume slider, but I don't even see that when nothing works. I don't even see a log in `journalctl`. Once it stops working, the only way I can find to change volume is on hifiberry.local. It never starts working again on spotify unless I start a new connection with a different device.

**HiFiBerryOS version**
20210203

**HiFiBerry sound card**
Amp2

**To Reproduce**
Begin playing via spotify using your phone. Try and change the volume. Sometimes it works, sometimes it doesn't. 

**Expected behavior**
The phone volume should reflect on the pi.

hifiberry, 2021-04-19:
Unfortunately that's not something we can reproduce.

tompeel, 2024-04-10:
Just noting here - sometimes on my android phone (pixel 7) android OS changes the in focus volume to be the phones media rarther than the Spotify specific volume.

You should see in the volume menu which pops up when you press volume up or down 2 volumes and you must select the Spotify or External media one... jsut a thought