hifiberry/hifiberry-os #258: Problem with NAS connecton after latest software update

milanmasat, 2021-04-16:
DO NOT remove the blocks below, but fill these with the requested data. Incomplete bug reports will be ignored!
You should remove this comment before posting the bug report.

**Describe the bug**
After the last Hifiberry update, music on the NAS (Synology DS215j) is no longer available. Nor did removing and adding the NAS again work. The problem was solved by reverting to the previous version of the software (20210203).
I tried updating due to issues with displaying currently playing music (http: //hifiberry.local/#now-playing), but it was worse than before.

**HiFiBerryOS version**
OS version 20210408

**HiFiBerry sound card**
DAC2 Pro

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
Available music from my NAS in Hifiberry :)

**Screenshots**
This is visible on version 20210203. On new version 20210408 is there visible conection problem:
[image]

On the NAS, the MUSIC folder is shared for reading on the internal network. I do not anticipate problems with the settings because everything worked on the previous version of the SW.
[image]

**Browser (if applicable)**
 - OS: W10
 - Browser Chrome - latest version
 - Version 89.0.4389.128

**Additional context**
No

akelge, 2021-04-17:
I can confirm, same issue, using plain samba

akelge, 2021-04-18:
Here is an excerpt from beocreate2 log, when I try to add a source (loki is my NAS)
```
Apr 18 08:32:56 the-musical-box beocreate2[728]: NAS configuration line 0 was not recognised:
Apr 18 08:33:09 the-musical-box beocreate2[728]: Added 'media/' from NAS 'loki'.
Apr 18 08:33:09 the-musical-box beocreate2[728]: NAS 'loki-media-mQg0f' does not appear to be mounted.
Apr 18 08:33:09 the-musical-box beocreate2[728]: NAS configuration line 1 was not recognised:
```

hifiberry, 2021-04-19:
@akelge Do you mean the upgrade has modified the /etc/smbmounts.conf file? This should not happen

hifiberry, 2021-04-19:
@milanmasat : What's the output of the command

 /opt/hifiberry/bin/mount-smb.sh

akelge, 2021-04-19:
> @akelge Do you mean the upgrade has modified the /etc/smbmounts.conf file? This should not happen

I mean that with older version I am able to specify a path, while with new version if I try to do it, I can't close the "Add NAS" page, the only way to add the NAS is without specifying the path, but then I am hit by the main bug

hifiberry, 2021-04-19:
Ok, but then you're talking about another bug - one in the UI. Did I understand this correctly? If yes, please create a separate bug report as this seems to be a different issue.

milanmasat, 2021-04-19:
> @milanmasat : What's the output of the command
> 
> /opt/hifiberry/bin/mount-smb.sh

hifiberry, 2021-04-19:
@milanmasat : Why did you close the bug report?

milanmasat, 2021-04-19:
> > @milanmasat : What's the output of the command
> > /opt/hifiberry/bin/mount-smb.sh

@hifiberry I replaced the password with ???. I shortened the library update.
# /opt/hifiberry/bin/mount-smb.sh
mountid=undefined-music-YHZtO, share=//192.168.0.102/music/0-HI-RES, user=milan,                                                                                                                                    passwd=???, mountopts=rw
mount error(16): Device or resource busy
Refer to the mount.cifs(8) manual page (e.g. man mount.cifs) and kernel log mess                                                                                                                                   ages (dmesg)
mountid=undefined-music-aDyy5, share=//192.168.0.102/music/1-FLAC, user=milan, p                                                                                                                                   asswd=???, mountopts=rw
mount error(16): Device or resource busy
Refer to the mount.cifs(8) manual page (e.g. man mount.cifs) and kernel log mess                                                                                                                                   ages (dmesg)
mountid=undefined-music-uRzA2, share=//192.168.0.102/music/9-Mluvene_slovo, user                                                                                                                                   =milan, passwd=???, mountopts=rw
mount error(16): Device or resource busy
Refer to the mount.cifs(8) manual page (e.g. man mount.cifs) and kernel log mess                                                                                                                                   ages (dmesg)
# INFO: root - return code 0 only if new covers were found
INFO: root - loaded processed files from /data/library/music/getcovers.log
INFO: root - Extracting covers from /data/library/music
# INFO: root - /data/library/music/undefined-music-aDyy5/Orient/Japanese Ceremonies
INFO: root - /data/library/music/undefined-music-aDyy5/Jazz/Ella Fitzgerald & Louis Armstrong/The Mag                                                                                                              ic Duets
INFO: root - /data/library/music/undefined-music-aDyy5/Pop Music/Celine Dion/Let´s Talk About Love
INFO: root - /data/library/music/undefined-music-aDyy5/Pop Music/Lenka
INFO: root - /data/library/music/undefined-music-aDyy5/Pop Music/Tom Jones/Greatests Hits
INFO: root - /data/library/music/undefined-music-aDyy5/Pop Music/Top Pop
INFO: root - /data/library/music/undefined-music-aDyy5/Pop Music/The Beach Boys/20 Great Love Songs
INFO: root - /data/library/music/undefined-music-aDyy5/Italská/AL BANO & ROMINA POWER - I GRANDI SUCC                                                                                                              ESSI/CD3
INFO: root - /data/library/music/undefined-music-aDyy5/Italská/AL BANO & ROMINA POWER - I GRANDI SUCC                                                                                                              ESSI/CD1
INFO: root - /data/library/music/undefined-music-aDyy5/Italská/AL BANO & ROMINA POWER - I GRANDI SUCC                                                                                                              ESSI/CD2
INFO: root - /data/library/music/undefined-music-aDyy5/Dětské/Písničky z večerníčků
INFO: root - /data/library/music/undefined-music-aDyy5/Vánoční/Nejkrásnější vánoční koledy (flac)
INFO: root - /data/library/music/undefined-music-aDyy5/Vánoční/Bílé Vánoce Lucie Bílé (flac)
INFO: root - /data/library/music/undefined-music-aDyy5/Vánoční/Ryba Česká mše vánoční (flac)
INFO: root - /data/library/music/undefined-music-aDyy5/Vánoční/Nebeské hlasy (Vánoční koledy a capell                                                                                                              a) (flac)
INFO: root - /data/library/music/undefined-music-aDyy5/Vánoční/Vánoční koledy - 29 nejznámějších kole                                                                                                              d (flac)
INFO: root - /data/library/music/undefined-music-aDyy5/Vánoční/Merry Christmas (flac)
INFO: root - /data/library/music/undefined-music-aDyy5/Soundtracks/The Twilight Saga - New Moon - The                                                                                                               Score
INFO: root - /data/library/music/undefined-music-aDyy5/Soundtracks/The Twilight Saga - New Moon - The                                                                                                               Score - výběr
INFO: root - /data/library/music/undefined-music-aDyy5/Soundtracks/The Twilight Saga - Twilight
INFO: root - /data/library/music/undefined-music-aDyy5/Soundtracks/Fifty Shades Of Grey Original Moti                                                                                                              on Picture Soundtrack
etc. ....

hifiberry, 2021-04-19:
According to this, the share seems to be mounted correctly. What's the specific problem?

milanmasat, 2021-04-19:
> @milanmasat : Why did you close the bug report? The comment above was for @akelge

milanmasat, 2021-04-19:
I am sorry @hifiberry . The bug report is a little bit confusing for me. 
For me, the problem was solved over the weekend by using an IP address. It didn't work without IP address. I had to enter it. In the previous version, it worked even without an IP address.

hifiberry, 2021-04-19:
Yes, that's correct. There is a bug when using the host name that was introduced in the April version.  We just reverted this and we'll probably have a bugfix release soon. As a workaround, using the IP address is just fine.

closed 2021-09-17 by hifiberry