hifiberry/hifiberry-os #259: SMB storage mounting fails if password contains special character & (solution found)

jn-simonnet, 2021-04-20:
When entering a password which contains a & character, we are unable to authenticate.
The solution is to escape the & with \, authentication succeeds and we are presented with the list of shares on the SMB server

hifiberry, 2021-04-21:
I don't fully understand your solution. Do you have a patch?

jn-simonnet, 2021-04-21:
No I do not have a patch.

What I signalled was that when you have a password which contains a & character, you need to escape it with a \ to fix the issue,
I mean, to use password titi&tata, you need to type titi\&tata in the GUI field. This was my quick fix I shared as others may also have encountered the problem.

Now in termes of patching, we would have to address the case of special characters in the shell script by escaping them in the script so that one just types the password.

By the way, I believe we he the issue also for \, which would have to be entered as \\ , " entered as \", ' as \' and possibly other characters having a special meaning for shell scripts.

Here is a discussion which I believe is relevant to solving it in the bash script.

[https://unix.stackexchange.com/questions/408974/convert-password-with-special-characters-for-use-with-expect-script](https://unix.stackexchange.com/questions/408974/convert-password-with-special-characters-for-use-with-expect-script)

If I have time, I will look further into the issue, I believe one has to modify the /opt/hifiberry/bin/mount-smb.sh script.

hifiberry, 2021-04-21:
Ok, thanks for the feedback.

jn-simonnet, 2021-04-21:
IN the script, we have this line
**mount.cifs "${SHARE}" "$BASEDIR/$MOUNTID" -o user="${USER}",password="${PASSWORD}","${MOUNTOPTS}"**

which surrounds password with " ", however the problem I believe occurs on this line :
**PASSWORD="${parts[3]//$'\n'}" # fixes newline inserts by Beocreate**

I just did not take the time to fix it yet.

jn-simonnet, 2021-04-21:
I quickly went looking into the script, and realized that we also have an issue if the password contains character ; which is used as the separator in file /etc/smbmounts.conf, but this is a different problem.

hifiberry, 2021-05-17:
If you have a patch, this would help.

closed 2021-11-08 by hifiberry