hifiberry/hifiberry-os #260: Disable WiFi option

MarcelRei, 2021-04-29:
**Is your feature request related to a problem? Please describe.**
After every reboot of my Pi4 HiFiBerry OS turns on WiFi. I want it to stay off, so nobody else could connect to it via WiFi

**Describe the solution you'd like**
Add an option to turn of WiFi on boot

**Describe alternatives you've considered**
Adding an option to set a WPA Key after setup so nobody without the key can connect

**Additional context**

hifiberry, 2021-04-30:
You mean it starts an access point? This happens if no network is connected

MarcelRei, 2021-04-30:
But my Pi is connected via ethernet ,with a manual set IP address, from the HifiBerry OS settings.
But I still see the HiFiBerry Setup Network broadcasted by the Pi.

hifiberry, 2021-04-30:
This is a sign that something isn't working correctly in your network and the system does not have access to the Internet. I would start checking this.

To disable the AP mode, you can create a noap file:
https://github.com/hifiberry/hifiberry-os/blob/master/doc/hacks.md

closed 2021-04-30 by hifiberry