hifiberry/hifiberry-os #261: BUG: Album has empty list of songs

MTxx87, 2021-05-05:
**Describe the bug**
One of the albums on the USB stick connected to the RP shows empty list of songs despite all files being there.

**HiFiBerryOS version**
20210408

**HiFiBerry sound card**
Hifiberry DIGI plus

**To Reproduce**
Steps to reproduce the behavior:
1. Go to http://hifiberry.local/
2. click Sources
3. click Music
4. select Artist 
5. select Album
6. BUG: view is empty (see screenshots below)

**Expected behavior**
The list of songs should be on the view.

**Screenshots**
[image]
[image]
[image]

**Browser (if applicable)**
 - OS: MacOS
 - Browser Safari (Version 14.1 ) / Chrome (Version 90.0.4430.93)

**Additional context**
Console error says:

`vue.js:634 [Vue warn]: Error in render: "RangeError: Invalid time value"
vue.js:1897 RangeError: Invalid time value
    at Vue.time (music-client.js:279)
    at eval (eval at createFunction (vue.js:11649), <anonymous>:3:6709)
    at Proxy.renderList (vue.js:2637)
    at eval (eval at createFunction (vue.js:11649), <anonymous>:3:6613)
    at Proxy.renderList (vue.js:2637)
    at eval (eval at createFunction (vue.js:11649), <anonymous>:3:6487)
    at Proxy.renderList (vue.js:2637)
    at Proxy.eval (eval at createFunction (vue.js:11649), <anonymous>:3:3178)
    at Vue._render (vue.js:3551)
    at Vue.updateComponent (vue.js:4067)`
    
As far as I know, the songs files are not corrupted and have all their meta in place.
I have compared them to the other FLAC files I have on the USB stick (which are all working).

hifiberry, 2021-05-06:
Usually this is a problem of missing/incorrect metadata. Check the metadata with an tag editor.

ssws, 2021-05-06:
Hi, thanks for the reply.

I have edited and checked the meta.
Here is a screenshot I have taken from MusicBrainz Picard app.

[image]

Please advise if you see something incorrect.
I couldn't find anything wrong.

Since the error in the console mentioned an error on the date, 
I tried both having the date set as year only (1991) or to a full date (1991-05-06), 
but no success.

If can help I can also share the file.

MTxx87, 2021-05-06:
Hi, thanks for the reply.

I have edited and checked the meta.
Here is a screenshot I have taken from MusicBrainz Picard app.

[image]

Please advise if you see something incorrect.
I couldn't find anything wrong.

Since the error in the console mentioned an error on the date, 
I tried both having the date set as year only (1991) or to a full date (1991-05-06), 
but no success.

If can help I can also share the file.

MTxx87, 2021-05-10:
As @hifiberry pointed out, it was a problem of metadata.

In case somebody else have the same problem:
manually changing metadata using MusicBrainz didn't work for me.
It started working perfectly only after I used the 'lookup' function and let the software assign the metadata for me.

closed 2021-05-10 by MTxx87