hifiberry/hifiberry-os #262: Can't change volume in snapcast

Sweenu, 2021-05-10:
**Describe the bug**
After some time, I'm not able to change the volume through snapcast unless I restart, after what it will work again for some time.
I don't know exactly what causes the volume change to stop working, so I cannot reproduce reliably. Even stranger, muting and unmuting always works, even when I can't change the volume.

**HiFiBerryOS version**
20210408

**HiFiBerry sound card**
Amp 2

**Additional context**
The output of `journalctl -u snapcastmpris`:
```
May 10 17:59:44 hifiberry systemd[1]: Starting Snapcast MPRIS...
May 10 17:59:45 hifiberry systemd[1]: Started Snapcast MPRIS.
May 10 17:59:48 hifiberry python3[432]: INFO: root - read /etc/snapcastclient.conf
May 10 17:59:48 hifiberry python3[432]: INFO: root - name on DBus aqcuired
May 10 17:59:48 hifiberry python3[432]: INFO: root - Obtained snapserver streaming port through zeroconf: 1704
May 10 17:59:48 hifiberry python3[432]: INFO: root - starting Snapclient
May 10 17:59:48 hifiberry python3[432]: INFO: root - snapclient now running in background
May 10 17:59:50 hifiberry python3[432]: INFO: root - Finding MAC address of active interface to use as snapclient id
May 10 17:59:50 hifiberry python3[432]: INFO: root - Status for interface docker0: down
May 10 17:59:50 hifiberry python3[432]: INFO: root - MAC address for interface docker0: 02:42:8a:5a:93:8a
May 10 17:59:50 hifiberry python3[432]: INFO: root - Status for interface wlan0: up
May 10 17:59:50 hifiberry python3[432]: INFO: root - MAC address for interface wlan0: e4:5f:01:03:4c:8b
May 10 17:59:50 hifiberry python3[432]: INFO: root - Status for interface dummy0: down
May 10 17:59:50 hifiberry python3[432]: INFO: root - MAC address for interface dummy0: e6:cf:0f:c2:aa:6a
May 10 17:59:50 hifiberry python3[432]: INFO: root - Status for interface eth0: down
May 10 17:59:50 hifiberry python3[432]: INFO: root - MAC address for interface eth0: e4:5f:01:03:4c:88
May 10 17:59:50 hifiberry python3[432]: INFO: root - Multiple MAC addresses, determining id
May 10 17:59:50 hifiberry python3[432]: INFO: root - Getting snapserver clients
May 10 17:59:50 hifiberry python3[432]: INFO: root - Client 3572c4a0-cba5-46cd-a36b-c4d41fe61fba is currently disconnected (was connected to Mopidy)
May 10 17:59:50 hifiberry python3[432]: INFO: root - Client 7f0df524-db2e-4e39-8a38-fdb2f59c4793 is currently disconnected (was connected to default)
May 10 17:59:50 hifiberry python3[432]: INFO: root - Client e4:5f:01:03:4c:8b is connected to stream Spotify
May 10 17:59:50 hifiberry python3[432]: INFO: root - Client c8:58:c0:cf:48:4c is currently disconnected (was connected to Spotify)
May 10 17:59:50 hifiberry python3[432]: INFO: root - Client e18ec52d-21cd-45a3-a526-84fe6c7083ec is currently disconnected (was connected to Mopidy)
May 10 17:59:50 hifiberry python3[432]: INFO: root - Client e79019b2-60b5-4fd0-9b99-fa9118b0158e is currently disconnected (was connected to Mopidy)
May 10 17:59:50 hifiberry python3[432]: INFO: root - Found mac address registered in snapserver: e4:5f:01:03:4c:8b
May 10 17:59:50 hifiberry python3[432]: INFO: root - Snapserver RPC version is 2.0.2
May 10 17:59:50 hifiberry python3[432]: INFO: root - Started SnapcastRpcWebsocketWrapper loop
May 10 17:59:50 hifiberry python3[432]: INFO: root - starting Snapclient
May 10 17:59:50 hifiberry python3[432]: ERROR: root - Snapcast RPC websocket error
May 10 17:59:50 hifiberry python3[432]: ERROR: root - Port could not be cast to integer value as 'e4:5f:01:03:4c:8b'
May 10 17:59:50 hifiberry python3[432]: INFO: root - Snapcast RPC websocket closed!
May 10 17:59:50 hifiberry python3[432]: INFO: root - Ending SnapcastRpcWebsocketWrapper loop
May 10 17:59:50 hifiberry python3[432]: INFO: root - snapclient now running in background
May 10 17:59:50 hifiberry python3[432]: INFO: root - ALSA <-> Snapcast volume synchronisation is disabled
May 10 17:59:50 hifiberry python3[432]: INFO: root - Snapcast wrapper thread started
May 10 17:59:52 hifiberry python3[432]: INFO: root - main loop started
```
There's also sometime two snapclient processes started by the service which I think is weird and could be the root cause.

rindlerblabla, 2022-01-08:
Did you ever get this working? Can't change the volume when using Snapcast either.

sweenu, 2022-01-08:
@rindlerblabla Sorry, no, it's still inconsistent.

rindlerblabla, 2022-01-10:
Have you tried changing the Snapclient options? Just SSH:d and changed the config file, adding `SNAPCLIENT_OPTS="--mixer hardware"`. So far it is workning, but don't know if it will be consistent.

Bertware, 2022-10-02:
```
May 10 17:59:50 hifiberry python3[432]: ERROR: root - Snapcast RPC websocket error
May 10 17:59:50 hifiberry python3[432]: ERROR: root - Port could not be cast to integer value as 'e4:5f:01:03:4c:8b'
```
It looks like there is some issue with the configuration, as the snapcast integration could not connect to the snapcast server, which is used for volume syncing

hifiberry, 2023-04-04:
Is this still active?

sweenu, 2023-04-04:
I still have this problem, even though I'm on the latest version of hifiberry-os, yes.