hifiberry/hifiberry-os #263: Click / pop sound when starting to play anything

j-lechner, 2021-05-13:
**Describe the bug**
There is an annoying click / pop sound every time I am switching to a new track via Spotify or even - as stated below - just when trying to play silence from the command line.

**HiFiBerryOS version**
20210408 running on RPI 4

**HiFiBerry sound card**
DAC2 HD + DSP ADD-ON BOARD

**To Reproduce**
(1) Login with ssh

(2) Run command
/usr/bin/aplay -D default -t raw -r 44100 -c 2 -f S16_LE /dev/zero

(3) A click sound can be heard right at the beginning

**Expected behavior**
No click sound, just silence.

j-lechner, 2021-05-13:
This must be an issue specific to HifiBerry OS since:
The click / pop sound does **NOT** occur when driving the DAC2 HD + DSP from the current DietPi image (5.10.17-v7+).

I've validated this by both playing silence via aplay and tracks via librespot.

However now I am of course missing all the goodness and convenience of HifiBerry OS, so I'd love to see this fixed.

j-lechner, 2021-05-13:
Manually upgrading HifiBerry OS 20210408 to **Kernel 5.10.17-v7+** did the trick:
I've just extracted packages raspberrypi-bootloader_1.20210430-2_armhf.deb and raspberrypi-kernel_1.20210430-2_armhf.deb and copied them into /boot and /lib/modules/

Can this be incorporated into a future HifiBerry OS release?

vonbremen-hauke, 2021-06-12:
I have the same problem. DAC2 HD clicks and pops before every song.
I can tell, that Kernel 5.10 is now merged in master, but no new release has been created...
Eagerly waiting the next release...

I am currently running 20210531...

vonbremen-hauke, 2021-06-12:
BTW, I would not consider this to be a feature-request nor medium prio...

hifiberry, 2021-06-14:
@j-lechner You might try the new experimental release with the latest 5.10 kernel:

/opt/hifiberry/bin/upgrade --experimental

j-lechner, 2021-06-14:
@hifiberry this release works for me, the click / pop sound is gone.

Thanks so much!

closed 2021-06-14 by j-lechner