hifiberry/hifiberry-os #265: API/Interact - Shutdown is mising

kongsted, 2021-05-16:
**Describe the bug**
The option to add the action for controling the power (shutdown / restart) is missing in the latest stable release.

**HiFiBerryOS version**
20210408 - is missing the option
20210203 - are having the option

**HiFiBerry sound card**
The system-info says "DAC+/Amp2", but I don't have amplifier on my hifiberry. :/
It is "only" a "DAC+ Standard".

**To Reproduce**
Steps to reproduce the behavior:
1. Open the webinterface
2. Click "Interact"
3. Click "New Interaction"
4. Click "Add Action"
5. Power is available in the old version - in the latest version "Power" isn't available

**Expected behavior**
Expected to have the option available :)
If calling the API/URL on the latest version it still answer "200 OK" even though it isn't going well.
(The version 20210203 is also answering "200 OK" even though it's an invalid API there's called)

**Screenshots**
From the latest version:
[image]

From version 20210203
[image]

**Additional context**
Using Firefox normally - have also tested in Edge Chromium

hifiberry, 2021-05-21:
Strange. I don't see this here with the latest version, but nothing has been changed with this. Try the latest experimental release

/opt/hifiberry/bin/update --experimental

kongsted, 2021-06-03:
It was there after I updated it some weeks ago with the example above (to version 20210520).
Now on version 20210531 it's gone again.... :/

reisschuessel, 2021-06-13:
I would also appreciate it very much if the "power" action would come back.
I am also using 20210531.

hifiberry, 2021-06-14:
It will ;-) This was a bug while refactoring some stuff in the UI. You can either wait for the next release or try the new experimental release:
/opt/hifiberry/bin/upgrade --experimental

hifiberry, 2021-07-22:
Did you test the experimental release?

reisschuessel, 2021-07-22:
> Did you test the experimental release?

Sorry, no i did not. I will wait untill the next regular release with this option.

reisschuessel, 2021-08-11:
I updated to the latest official release and the "Power" option is available (sort of).
But when selected i can not save the interaction. Either Web Request or Serial Message have this issue.

**HiFiBerryOS version**
20210804
I tested on two different HiFiBerry Systems with this release.

**To Reproduce**
Steps to reproduce the behavior:

    Open the webinterface
    Click "Interact"
    Click "New Interaction"
    Click "Add Action"
    Click "Power"
    Click "Shut Down Raspberry Pi'"
    Click "Save Action"

**Expected behavior**
Expected to have the action to show up under "DO" and have the button "Save Interaction" available.
As it is now the action is not shown. Using e.g. one of the "Set Volume" actions it works as intended.

reisschuessel, 2021-11-23:
Still not available/working in latest 20211105 release.

StephenBeirlaen, 2021-12-03:
Also having this problem on 20211105.
I've already tried setting another trigger action, and then fiddling with `/etc/beocreate/interact.json` but couldn't find the right values unfortunately.

StephenBeirlaen, 2021-12-03:
Never mind, I found the expected values by looking at the serverside.
You can bypass the bug by editing `/etc/beocreate/interact.json`:

`
{"name":"Turn off HTTP","triggerData":{"addressEnd":"turn-off"},"actions":[{"extension":"general-settings","type":"power","data":{"option":"shutdown"}}],"triggerExtension":"interact","triggerType":"httpAPI"}
`

astremtan, 2022-02-06:
I know this is old but I still have this issue with my hifiberry on 20211105.
In case anyone is looking, the expected value for restarting is {"option":"restart"}.
After editing the interact.json file, a manual restart is required for changes to take effect.

hifiberry, 2022-08-04:
This has been fixed in 20220726

closed 2022-08-04 by hifiberry