hifiberry/hifiberry-os #267: Bluetooth not showing up

mribrgr, 2021-05-20:
DO NOT remove the blocks below, but fill these with the requested data. Incomplete bug reports will be ignored!
You should remove this comment before posting the bug report.

**Describe the bug**
Bluetooth is not showing up after installing HifiBerryOS latest version on RPI4 4GB with Amp2 1.0.

**HiFiBerryOS version**
20210520

**HiFiBerry sound card**
Amp2 1.0

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Sources'
2. Click on 'Disabled Sources'
3. Airplay 1, Music, Radio, DLNA, OpenHome, Snapcast and Sqeezelite showing up, but no Bluetooth

**Expected behavior**
Bluetooth is showing up.

**Screenshots**
[image]

**Browser (if applicable)**
- every Browser i tried, newest iOS with Safari, Firefox and Chrome newest version.

**Additional context**
-

hifiberry, 2021-05-21:
I don't see this here. Are you sure you're using a Pi4. Please post the output of

cat /etc/hifiberry.state

mribrgr, 2021-05-21:
@hifiberry there is no /etc/hifiberry.state. In /etc/ is only the hifiberry.version (output: 20210520) and the hifiberry.conf.sample and the hifiberry folder.
I installed hifiberryos using the latest image from https://www.hifiberry.com/images/hifiberryos-pi4.zip
and wrote it using etcher.

mribrgr, 2021-05-21:
And yes, I am using a Pi4 with the pi4 image. I tested it before with the pi3b, but read, that there might be issues with BT, so I bought a Pi4 4GB.

hifiberry, 2021-05-21:
Ok, what's the output of 

/opt/hifiberry/bin/reconfigure-players

mribrgr, 2021-05-21:
# cat /opt/hifiberry/bin/reconfigure-players
#!/bin/bash

REBOOTFILE=/tmp/reboot

SERVICES="spotify raat mpd shairport-sync squeezelite bluealsa-aplay alsaloop dlnampris mopidy upmpdcli"

FEATURES="pi"

# Check that UUID exists
if [ ! -s "/etc/uuid" ]; then
 /usr/bin/uuidgen > /etc/uuid
fi

if [ ! -d /etc/features ]; then
 # Looks like a fresh installation or update, 
 # reconfigure everything
 if [ -f /etc/hifiberry.state ]; then
   mv /etc/hifiberry.state /etc/hifiberry.state.bak
 fi
 mkdir /etc/features
fi 

read_features() {
 for dir in /etc/hifiberry /custom/hifiberry; do
   if [ -d $dir ]; then 
     pushd $dir
     for f in *.feature; do
       if [ "$f" != "*" ]; then
         F=${f/.feature/}
         FEATURES="$FEATURES $F"
       fi 
     done
   popd
   fi
 done
}

restart_service() {
 systemctl is-active --quiet $1
 if [ "$?" == "0" ]; then
   echo "Restarting $1"
   systemctl restart $1
 fi
}

read_fixed() {
 if [ -f /etc/hifiberry/config.fixed ]; then
  . /etc/hifiberry/config.fixed
  for N in MIXER_CONTROL CARD FEATURES; do
    $N=${CURRENT_$N}
  done
 fi
}

check_systemd() {
 SERVICE_ACTIVE=0
 for SERVICE in "$@"; do 
   systemctl is-active --quiet $SERVICE
   if [ "$?" == "0" ]; then
     SERVICE_ACTIVE=1
   fi

   systemctl is-enabled --quiet $SERVICE 
   if [ "$?" == "0" ]; then
     SERVICE_ACTIVE=1
   fi 
   
 done
}
[... 679 lines cut ...]
reconfigure_vollibrespot
echo "Reconfiguring Shairport-sync"
reconfigure_shairport
echo "Reconfiguring Bluetooth"
reconfigure_bluetooth
echo "Reconfiguring squeezelite"
reconfigure_squeezelite
echo "Reconfiguring MPD"
reconfigure_mpd
echo "Reconfiguring DLNA"
reconfigure_dlna
echo "Reconfiguring Mopidy"
reconfigure_mopidy
echo "Reconfigure AudioControl2"
reconfigure_audiocontrol2
echo "Reconfigure upmpdcli"
reconfigure_upmpdcli
echo "Reconfigure RAAT"
reconfigure_raat
echo "Reconfiguring Beocreate software"
reconfigure_beocreate

if [ "$1" == "--reboot" ]; then
  echo "System configuration changed" >> $REBOOTFILE
fi

hifiberry, 2021-05-21:
not cat, just run the command

/opt/hifiberry/bin/reconfigure-players

mribrgr, 2021-05-21:
Maybe this is also interesting:
# tr -d '\0' < /proc/device-tree/model
Raspberry Pi 4 Model B Rev 1.1

mribrgr, 2021-05-21:
I get the error:

# /opt/hifiberry/bin/reconfigure-players
/opt/hifiberry/bin/reconfigure-players: line 374: syntax error near unexpected token `}'
/opt/hifiberry/bin/reconfigure-players: line 374: `}'

hifiberry, 2021-05-21:
Ok, this was a bug that has been fixed now. Try updating the system 

/opt/hifiberry/bin/update --experimental

mribrgr, 2021-05-23:
When trying to update the system this way it downloads the updater-20210521-pi0w.zip, which isn't the correct version for my pi4. Afterwards my system does not work anymore (nothing works when trying to power it) and I need to reinstall the os.

hifiberry, 2021-06-04:
Please try again the latest release.

hifiberry, 2021-06-14:
Did you test the latest release?

closed 2021-11-02 by hifiberry