hifiberry/hifiberry-os #268: incorrect covert art shown for some tracks

t-alonso, 2021-05-25:
I'm using the OpenHome/DLNA source to stream from a DLNA server, which works fine, but the GUI displays the wrong cover art for some tracks.
(upmpdcli   901     1  0 May20 ?        00:29:17 /bin/upmpdcli -c /etc/upmpdcli.conf)

Here are 2 recent examples (metadata from the :81/api/track/metadata):
{"artist": "Rhye", "title": "Fire", "albumArtist": "Rhye", "albumTitle": "Home", "artUrl": null, "externalArtUrl": "http://coverartarchive.org/release/74686b21-79c6-4274-8fb0-e2db1ca02fa9/16336559290.jpg", "discNumber": "1", "tracknumber": "11", "playerName": "mpd", "playerState": "playing", "streamUrl": "http://192.168.0.32:9000/disk/DLNA-PNMP3-OP01-FLAGS01700000/O0$1$8I2679563.mp3", "playCount": null, "mbid": "1c8df792-76ba-4448-9c0d-68998a46e366", "artistmbid": "aa914efa-d6cb-4816-906f-8edafb08418e", "albummbid": "4eacb63d-31f3-47b0-8840-4148b4b3e160", "loved": null, "wiki": null, "loveSupported": false, "tags": [], "skipped": false, "host_uuid": "00bea6ea-84f1-4976-9f16-b0a7fdaaa8b4", "releaseDate": "2017-09-01", "trackid": null, "hifiberry_cover_found": true, "duration": 0, "time": 0, "position": 0, "positionupdate": 1621949540.5435462}

where the cover art shown is for "WizKid - Come Closer ft. Drake" . . . 

Another example (:81/api/track/metadata):
{"artist": "Original Broadway Cast of Hamilton", "title": "Yorktown (The World Turned Upside Down)", "albumArtist": "Original Broadway Cast of Hamilton", "albumTitle": "Hamilton (Original Broadway Cast Recording)", "artUrl": null, "externalArtUrl": "http://coverartarchive.org/release/6dc16cc8-41aa-4e3b-b440-fc9e194d6ac7/25072845970.jpg", "discNumber": "1", "tracknumber": "20", "playerName": "mpd", "playerState": "playing", "streamUrl": "http://192.168.0.32:9000/disk/DLNA-PNMP3-OP01-FLAGS01700000/O0$1$8I2304011.mp3", "playCount": null, "mbid": "567a4f52-497b-4fe9-acb6-34d0effd53a3", "artistmbid": "925c7673-0e85-410f-b7e4-d9705a7aa619", "albummbid": "093941fe-5416-471b-8548-86d9b8724a1d", "loved": null, "wiki": null, "loveSupported": false, "tags": ["musical", "alexanderhamiltonmusical"], "skipped": false, "host_uuid": "00bea6ea-84f1-4976-9f16-b0a7fdaaa8b4", "releaseDate": "2017-06-30", "trackid": null, "hifiberry_cover_found": true, "duration": 0, "time": 0, "position": 0, "positionupdate": 1621945839.4443572}

where the cover art shown is for the "Hadestown  musical ".

I'm happy to debug, let me know what to look at.

hifiberry, 2021-05-25:
This can happen if the source doesn't natively support cover art. In this case the system tries to find cover art on the Internet. This doesn't work perfectly, but there's nothing we can do about it.

t-alonso, 2021-05-26:
But I think it can work better.
I added the the artist and title from both of the above examples to the 
/opt/audiocontrol2/ac2/test_metadata.py
file and when run it does get much better results:

example 1:
{"artist": "Rhye", "title": "Fire", "albumArtist": null, "albumTitle": null, "artUrl": null, "externalArtUrl": "https://lastfm.freetls.fastly.net/i/u/300x300/a4f41b6c030df3a7abd641e2c93d40f3.png", "discNumber": null, "tracknumber": null, "playerName": null, "playerState": "unknown", "streamUrl": null, "playCount": null, "mbid": "fd77d067-c9fc-46ef-8fda-293256f477b0", "artistmbid": "aa914efa-d6cb-4816-906f-8edafb08418e", "albummbid": null, "loved": null, "wiki": null, "loveSupported": false, "tags": [], "skipped": false, "host_uuid": "00bea6ea-84f1-4976-9f16-b0a7fdaaa8b4", "releaseDate": null, "trackid": null, "hifiberry_cover_found": true, "duration": 0, "time": 0, "position": 0, "positionupdate": 1621998347.875987}

example 2:
{"artist": "Original Broadway Cast of Hamilton", "title": "Yorktown (The World Turned Upside Down)", "albumArtist": null, "albumTitle": null, "artUrl": null, "externalArtUrl": "http://coverartarchive.org/release/19e87c02-28df-4a89-ba7e-8f5ef474c36f/13308044887.jpg", "discNumber": null, "tracknumber": null, "playerName": null, "playerState": "unknown", "streamUrl": null, "playCount": null, "mbid": "fd126cd9-c40e-43f6-a3ca-68c68c1f682f", "artistmbid": "a0ef7e1d-44ff-4039-9435-7d5fefdeecc9", "albummbid": "19e87c02-28df-4a89-ba7e-8f5ef474c36f", "loved": null, "wiki": null, "loveSupported": false, "tags": ["musical", "alexanderhamiltonmusical"], "skipped": false, "host_uuid": "00bea6ea-84f1-4976-9f16-b0a7fdaaa8b4", "releaseDate": "2015-09-25", "trackid": null, "hifiberry_cover_found": true, "duration": 0, "time": 0, "position": 0, "positionupdate": 1621998006.5125275}

Another clue of what is going on is the fact that the "/#now-playing" GUI  says it is "RADIO" playing, but i'm playing to the OpenHome UPnP/AV renderer (OpenHome/DLNA).

hifiberry, 2021-05-26:
It's normal that it's displayed as a radio station as we can't distinguis between different streaming sources. Every stream is shown as "Radio". If you have patch to improve it, feel free to send it. We'll have a look.

t-alonso, 2021-05-29:
No proper patch because it is a very ugly hack, but I just added the below code to /opt/audiocontrol2/ac2/metadata.py and it does the trick for me (Twonky will serve the cover art from a URL similar to the music stream URL):

        if "/DLNA-PNWMABASE-OP01-FLAGS01700000/" in metadata.streamUrl:
            metadata.externalArtUrl=metadata.streamUrl.replace("/DLNA-PNWMABASE-OP01-FLAGS01700000/","/DLNA-PNJPEG_TN-OP01-CI1-FLAGS00d00000/defaa/A/").replace(".wma",".jpg")
        if "/DLNA-PNMP3-OP01-FLAGS01700000/" in metadata.streamUrl:
            metadata.externalArtUrl=metadata.streamUrl.replace("/DLNA-PNMP3-OP01-FLAGS01700000/","/DLNA-PNJPEG_TN-OP01-CI1-FLAGS00d00000/defaa/A/").replace(".mp3",".jpg")

hifiberry, 2021-06-04:
If you can provide a less "hacky" method for this, we're happy to include it.

closed 2022-07-21 by hifiberry