hifiberry/hifiberry-os #269: Roon won't enable

wtfuzz, 2021-05-25:
**Describe the bug**
Enabling the Roon source via the web UI doesn't enable the source. Going back to the sources page shows Roon under the disabled sources list. The roon bridge does not appear as an audio sink on my core.

**HiFiBerryOS version**
20210521

**HiFiBerry sound card**
Amp2

**To Reproduce**
1. Go to sources
2. Expand Disabled Sources
3. Click on Roon
4. Click the on checkbox
5. Wait for "Turning roon on" to finish and redirect
6. Click on sources
7. Note that Roon appears under Disabled sources

**Expected behavior**
The Roon source to be enabled successfully.

**Screenshots**

**Browser (if applicable)**
- Chrome

**Additional context**

Divadeer, 2021-05-26:
Im having this exact same issue on the digi+pro on a pi3b+ I got today. Running systemctl start raat.service yields the following error when I check status.

 raat.service - Roon
     Loaded: loaded (/usr/lib/systemd/system/raat.service; enabled; vendor preset: enabled)
     Active: activating (auto-restart) (Result: exit-code) since Wed 2021-05-26 01:13:03 CDT; 1s ago
    Process: 2705 ExecStartPre=/opt/hifiberry/bin/bootmsg Starting Roon (code=exited, status=0/SUCCESS)
    Process: 2709 ExecStart=/opt/raat/raat_app /etc/hifiberry_raat.conf (code=exited, status=1/FAILURE)
   Main PID: 2709 (code=exited, status=1/FAILURE)

Thrilled to find out its just a software issue.

hifiberry, 2021-05-26:
Can you try the latest experimental:
/opt/hifiberry/bin/update --experimental

Divadeer, 2021-05-27:
> Can you try the latest experimental:
> /opt/hifiberry/bin/update --experimental

Works! Thanks.

wtfuzz, 2021-05-27:
This did not work for me. It updated to `20210525` and rebooted. Verified the version in web UI after the reboot.

```
● raat.service - Roon
     Loaded: loaded (/usr/lib/systemd/system/raat.service; enabled; vendor preset: enabled)
     Active: activating (auto-restart) (Result: exit-code) since Wed 2021-05-26 22:02:05 PDT; 4s ago
    Process: 932 ExecStartPre=/opt/hifiberry/bin/bootmsg Starting Roon (code=exited, status=0/SUCCESS)
    Process: 936 ExecStart=/opt/raat/raat_app /etc/hifiberry_raat.conf (code=exited, status=1/FAILURE)
   Main PID: 936 (code=exited, status=1/FAILURE)

```

Trying to start manually:

```
# /opt/raat/raat_app /etc/hifiberry_raat.conf
/opt/raat/raat_app: /lib/libasound.so.2: no version information available (required by /opt/raat/raat_app)
/opt/raat/raat_app: /lib/libasound.so.2: no version information available (required by /opt/raat/raat_app)
Invalid config /etc/hifiberry_raat.conf:-1: unable to open /etc/hifiberry_raat.conf: No such file or directory

```

hifiberry, 2021-05-27:
What's the output of

/opt/hifiberry/bin/reconfigure-players

wtfuzz, 2021-05-27:
```
# /opt/hifiberry/bin/reconfigure-players
/etc/hifiberry ~
~
Nothing changed, exiting...

```

hifiberry, 2021-05-27:
Ok, try

rm /etc/hifiberry.state

and run again

wtfuzz, 2021-05-27:
```
# rm /etc/hifiberry.state 
# /opt/hifiberry/bin/reconfigure-players
/etc/hifiberry ~
~
Sound card settings have been changed, reconfiguring...
Applying settings: 
 Players:  (was )
 Raspberry Pi model: 4
 System name: BackYard (was )
 Sound card: DAC+/Amp2 (was )
 Interrupt pin for power controller: 2 (was )
 DSP type:  (was )
 HW device: card 0 (was )
 ALSA mixer control: Digital (was )
 Recording:  (was )
 Vendor/product HiFiBerry DAC+ (was )
 Features: pi bluetooth pi3orlater arm7 localui (was )

Saving settings to /etc/hifiberry.state
Reconfigure features
Reconfiguring Spotifyd
 updated mixer
 updated system name
Restarting spotify
Reconfiguring Vollibrespot
 updated mixer
 updated system name
 updated mixer-card
Restarting spotify
Reconfiguring Shairport-sync
 updated mixer
 updated system name
Restarting shairport-sync
Reconfiguring Bluetooth
 updated system name
 updated mixer
Restarting bluetooth.service
Reconfiguring squeezelite
 updated system name
Reconfiguring MPD
Restarting mpd
 updated mixer
Reconfiguring DLNA
Reconfiguring Mopidy
modidy not installed, ignoring
Reconfigure AudioControl2
 updated mixer
 updated interrupt pin
Restarting audiocontrol2
Reconfigure upmpdcli
 updated system name
Reconfigure RAAT
RAAT not installed, ignoring
Reconfiguring Beocreate software
Value could not be interpreted as JSON, falling back to old method.
Settings updated for 'system'.
 updated card type
Value could not be interpreted as JSON, falling back to old method.
Settings updated for 'sound'.
 updated mixer
["pi","bluetooth","pi3orlater","arm7","localui"]
Settings updated for 'system'.
Restarting beocreate2
 restarting Beocreate

```

hifiberry, 2021-05-27:
Now reboot and try if it works

wtfuzz, 2021-05-27:
Still not working.

Not sure if you noticed `reconfigure-players` displayed ```RAAT not installed, ignoring```

hifiberry, 2021-05-27:
Ok, I need to look in this further.

wtfuzz, 2021-05-27:
I ran `/opt/raat/configure-raat` and it generated an `/etc/hifiberry_raat.conf` file for me, and the source has successfully started now

hifiberry, 2021-05-27:
That's a workaround, but we still have to figure out why it's not working as it should.

wtfuzz, 2021-05-27:
I'm happy to reinstall and test the proper solution

Divadeer, 2021-05-27:
I had actually installed the previous version last night to get it working and then ran the update command mentioned above this morning. You may want to try that (updating from the previous release to experimental instead of the current release to experimental). You wouldn't expect it would make a difference but sometimes in computing paradoxical things like that can work.

hifiberry, 2021-05-27:
New experimental is up that should fix this now.

wtfuzz, 2021-05-27:
* Wrote image to a fresh SD card with `20210521`
* Verified Roon will not enable successfully
* Ran experimental update to `20210527`
* Enabled Roon source and verified it started succesfully

Thanks for the quick fix!

jacobacci, 2021-06-03:
@hifiberry 
Just updated to 20210531. Roon is working again, thank you.

hifiberry, 2021-06-04:
Thanks for the feedback. Closing this

closed 2021-06-04 by hifiberry

delorean02, 2025-05-01:
Just had to re-image as my SD card failed.  Roon will not start, and RAAT is not installed on 20240523.  Any ideas?