hifiberry/hifiberry-os #273: Spotify stops playing after a few minutes

stenub, 2021-06-21:
**Describe the bug**
When I listen to songs via spotify connect, the playback stops every few minutes with an "os error 104" and the spotify service restarts.
This is no wifi problem and also no problem of my internet connectivity. Everything else is working fine.

**HiFiBerryOS version**
20210531

**HiFiBerry sound card**
DAC+ DSP on a Raspi 4B

**To Reproduce**
Play songs via spotify and wait a few minutes

**Expected behavior**
no interruption of playback

**Additional context**
output of journalctl -b -u spotify.service after an interruption:

Jun 21 09:54:09 hifiberry vollibrespot[759]: [Vollibrespot] : SessionError: Connection reset by peer (os error 104)
Jun 21 09:54:09 hifiberry systemd[1]: spotify.service: Succeeded.
Jun 21 09:54:14 hifiberry systemd[1]: spotify.service: Scheduled restart job, restart counter is at 1.
Jun 21 09:54:14 hifiberry systemd[1]: Stopped Vollibrespot.
Jun 21 09:54:14 hifiberry systemd[1]: Starting Vollibrespot...
Jun 21 09:54:14 hifiberry vollibrespot[1338]: vollibrespot v0.2.2 UNKNOWN UNKNOWN (librespot 1a224a3 2020-10-25) -- Built On 2021-05-19
Jun 21 09:54:14 hifiberry vollibrespot[1338]: Reading Config from "/etc/vollibrespot.conf"
Jun 21 09:54:14 hifiberry vollibrespot[1338]: [Vollibrespot] : Using Alsa backend with device: default

hifiberry, 2021-07-22:
"Connection reset by peer" means that the other side has terminated the connection. This often happens with unstable WiFi, but it might also be a problem with your firewall settings.

stenub, 2021-08-11:
Thanks for your response, but as i already stated, it is NOT the Wifi. I have the same errors when i connect via ethernet cable. 
Also, the internet connection is fine, no disconnects in the logs of the router.

hifiberry, 2021-08-11:
Nothing we can do about it as we can't reproduce this.

stenub, 2021-08-30:
Ok, now i watched the whole problem more closely. I connected the hifiberry with am Ethernet cable, so Wifi can't be the problem anymore. How can i further investigate what exactly the problem is? Is there any kind of debug mode in vollibrespot or hifiberry?
From my understanding, "connection reset by peer" means that the other side (spotify) closed the connection..
My internet is fine... no PPPoE disconnects, no vpn reconnects, nothing i am aware of. I am working from home all day, so i would recognize a problem when there would be one...

charles-waite, 2021-09-13:
Also seeing a similar problem. Playing audio via Spotify Connect seems to just stop after 5 minutes or so. 

Pi 3B connected via Wifi. Nothing crazy on my wifi setup, or firewall rules. Not sure how to do any logging though.

stenub, 2022-01-24:
> I have the same issue, i think it is due to i have Bluetooth devices that take over the output.

Try using 5GHz wifi. If it is really bluetooth then 5GHz should solve your problem.

stenub, 2022-02-06:
Hi, just a little update:
I used a spare SD Card and installed Volumio on it. Next I installed the Volumio Spotify Connect2 Plugin. As far as I have seen, volumio uses vollibrespot for Spotify Connect. Not a single disconnect so far. Ergo: It is definitly not the Wifi.

fredrike, 2022-02-07:
>  If it is really bluetooth then 5GHz should solve your problem.

The problem is that Bluetooth connections take over the audio output.

stenub, 2022-03-11:
Hey all,

i recently had time to further debug my connection problem and this is what i found out:
I made a support ticket at spotify and they gave me a temporary test account (maybe with debugging features activated).
I tested with this account for over an hour, and had no single disconnect. 
So it is **definitly not a network issue!**
Spotify seems to randomly drop the connection. Perhaps a kind of security measure, as Hifiberry/vollibrespot is not officially supported by spotify. 

Now the story goes on: I reported back and told them that it all works fine now. They made the proposal to create a completely new (production) account for me, i can further test with it for 24h, and if everything is all right, they would move all my playlists and so on from my old to the newly created account. I consented.
5 minutes later I had a new account, with enabled premium features and a tried to connect with the hifiberry... No connection was possible at all. Hifiberry wasnt even listed as a possible external playback device in the iOS app and desktop client. Also a reboot of the hifiberry changed nothing with this.
The support repeadetly claimed that "we have no cooperation with hifiberry" as the excuse for why it is not working anymore.
I told the spotify support, they should move me back to my old account. Now spotify connect is working again ( but still with interruptions).

It seems that spotify is planning to block connections with Hifiberry/vollibrespot in the near future or perhaps with newly created accounts, so it could be that vollibrespot will not work with spotify anymore..

I hope this is a "heads up" for every developer who is working on Hifiberry and vollibrespot.

P.S.: I did the same with Volumio, which also uses vollibrespot, with the exactly same results.

fredrike, 2022-03-11:
> Hey all,
>
> i recently had time to further debug my connection problem and this is what i found out: I made a support ticket at spotify and they gave me a temporary test account (maybe with debugging [...]

Nice find!

closed 2022-03-11 by hifiberry