hifiberry/hifiberry-os #274: Optical output does not turn off when playback stops

odel4y, 2021-07-11:
With a recent automatic update (I cannot say when exactly) the digital optical output does not automatically turn off anymore when music playback has stopped. Instead the Toslink LED always stays on. This used to work before and my speakers reacted to this by going to/waking from Standby. A fresh reinstall of 20210531 did not solve this problem.

Hardware: Raspberry Pi 3B+, Hifiberry Digi+ I/O

hifiberry, 2021-07-11:
You can find older releases here:
https://github.com/hifiberry/hifiberry-os/releases

There won't be any low-level driver changes by us. We use the default Raspberry Pi Linux kernel without modifications.

closed 2021-07-11 by hifiberry

moritzmhmk, 2021-10-17:
Any hint what causes this new behaviour? It breaks my HiFi setup that shares a digital port via a reverse splitter.

moritzmhmk, 2021-10-17:
TL,DR: To fix this one must set the `pmdown_time` option of the `snd_soc_core` module to 5000.

```bash
mkdir /etc/modprobe.d/
mv /etc/modules-load.d/snd_soc_core_disable_pm.conf /etc/modprobe.d/
echo "options snd_soc_core pmdown_time=5000" > /etc/modprobe.d/snd_soc_core_disable_pm.conf
```
----

I did some research and found a potential bug on the way:

The `pmdown_time` option of the `snd_soc_core` module defines how long the digital output will stay active when playback stops - the default is 5000 (i.e. 5 seconds). Setting this to -1 leaves the output on indefinitely.

The file `snd_soc_core_disable_pm.conf` should not be in `/etc/modules-load.d`  but in `/etc/modprobe.d` instead - it has no effect otherwise (besides an error in dmesg). (Note: the file was created with commit bdfde92a30099de7527987c58effe73d975a5eba)

One might wonder how the `pmdown_time` was changed to `-1` amid the file being in the wrong location - in fact there is another commit (349108c58e13dd3c4ababeef72e4ea79848f3776) that introduces a patch to hardcode `-1` as the default value.

(I am not sure if fb91eb59765ab27a75ce29694cde240ae22d8c19 is also related.)

I suggest that the hard coded value should be removed and the location of `snd_soc_core_disable_pm.conf` should be fixed making the setup more obvious and easier to manipulate.

It would also be awesome to have a toggle in the advanced sound settings to en-/disable standby of digital output.

yubiuser, 2021-10-20:
I'm affected by that bug as well. 
Thanks @moritzmhmk for the workaround. Hope this get's fixed in Hifiberry soon.

odel4y, 2021-11-01:
Thanks @moritzmhmk, this fix works for me in 20210804 as well!
Linking @hifiberry in case this hasn't been noticed.

yubiuser, 2021-11-02:
https://github.com/hifiberry/hifiberry-os/commit/2e70e4b2550d38d401b8e7d0fe3e08cbd9f7347e alone is not enough to fix the issue. I tried it, as long `options snd_soc_core pmdown_time` is set to `-1` the optical output won't shut down even when the file is in the right directory.

hifiberry, 2021-11-02:
That's the idea. If you want to change the behavior, you need to edit the file. As this impacts ALL soc power management and most user prefer to disable power management, we won't change this default setting

moritzmhmk, 2021-11-02:
The patch introduced with 349108c58e13dd3c4ababeef72e4ea79848f3776 should also be removed since the file (now in the correct position) already disables pm.

hifiberry, 2021-11-02:
@moritzmhmk : You're right, we'll remove this

hifiberry, 2021-11-02:
We try not to end up like other distributions that have a lot of options and settings. I don't think we will add this to the UI. 
I know that not everybody agrees with this, but we can't make it right for everyone. 
If somebody wants to change this, command line is the way to go. I'll make sure, this configuration file is preserved during system updates.
@yubiuser: There are lots of Digi users that want power management disabled because their DACs also create pops when resyncing. Therefore, there is no "right" way to do it and the decision is to leave it disabled by default.

odel4y, 2021-12-16:
I would second this. All other devices with an optical output that I own turn the diode off as soon as they go to standby. So this is somewhat expected behavior. Hence my suspicion of a bug when it suddenly stopped working. 

The pops that other users experience are a property of their DAC. Leaving the optical output turned on is, as I understand, a workaround for that specific issue. I find it is justified to offer that workaround. But I would also favor it as an advanced audio option as suggested by @moritzmhmk.

I fully support your goal of keeping the UI clean and tidy. However, in my opinion this setting would not be too technical for users to understand and it is too essential to leave it out.