hifiberry/hifiberry-os #275: Airplay 2 support

Smith8154, 2021-07-21:
DO NOT remove the blocks below, but fill these with the requested data. Incomplete feature requests will be closed.
You should remove this comment before posting the feature request.

**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
Not related to a problem.
**Describe the solution you'd like**
A clear and concise description of what you want to happen.
Airplay 2 support with shairport sync, as it now has airplay 2 support. https://github.com/mikebrady/shairport-sync/blob/development/AIRPLAY2.md
**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.
NA
**Additional context**
Add any other context or screenshots about the feature request here.
NA

hifiberry, 2021-07-22:
Coming: 
https://github.com/hifiberry/hifiberry-os/commit/054a37d342cb09a08e2b0eb7f0cedc4a0d99ae65

philaub, 2021-07-22:
> DO NOT remove the blocks below, but fill these with the requested data. Incomplete feature requests will be closed.
> You should remove this comment before posting the feature request.
> 
> **Is your feature request related to a problem? Please describe.**
> A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
> Not related to a problem.
> **Describe the solution you'd like**
> A clear and concise description of what you want to happen.
> Airplay 2 support with shairport sync, as it now has experimental airplay 2 support. Maybe a checkbox to enable the experimental feature, or add it to a beta version of hifiberry OS. https://github.com/mikebrady/shairport-sync/blob/development/AIRPLAY2.md
> **Describe alternatives you've considered**
> A clear and concise description of any alternative solutions or features you've considered.
> NA
> **Additional context**
> Add any other context or screenshots about the feature request here.
> NA

Same here.
A general guideline for the implementation of Airplay 2 would be extremely satisfying. Even though getting it to run stable maybe a  process.

Support for Airplay 2 is the most sought after feature missing in my opinion.

hifiberry, 2021-07-22:
Duplicate of https://github.com/hifiberry/hifiberry-os/issues/225

closed 2021-07-22 by hifiberry