hifiberry/hifiberry-os #276: Media skips every 6 to 7 seconds after reboot. Stops after a few hours.

js521458741, 2021-07-26:
purchased a HiFiBerry DAC+ ADC Pro. After install I used the Analog input and noticed an interruption in the music every 6 to 7 seconds. Like a skip loss. I disassembled everything and to make sure I hand't made a mistake. Put everything back together and got the same weird skip noise every 6 to 7 seconds. I looked online and couldn't find anything specific to my problem, so i took the HiFiBerry DAC+ ADC Pro off the Pi 4 and placed it on a different Pi4 and had the same result. The weird thing is that it doesn't happen right away. I changed the source devices (iPad,w/lightning, iPad/USB-C, iPhone, 2nd iPhone) and they all skipped. I changed the Pi4 and it skipped again.

It usually happens at boot. I left it alone, went for a 30 min run made dinner, etc. Came back a few hours later and the skip was gone. Reboot the device the next day and the skip is back. I'm at a loss. Any help?

js521458741, 2021-07-26:
Pi4 (2Gb + 4GB)
3.5mm input and RCA output.
hifiberryos-20210531

hifiberry, 2021-07-26:
We provide templates for bug reports:
https://github.com/hifiberry/hifiberry-os/issues/new/choose

Please use these.

closed 2021-07-26 by hifiberry