hifiberry/hifiberry-os #278: Music skips every 6-7 seconds even after rebooting.

js521458741, 2021-07-26:
DO NOT remove the blocks below, but fill these with the requested data. Incomplete bug reports will be ignored!
You should remove this comment before posting the bug report.

**Describe the bug**
Sound or Skip Loss every 6 - 7 Seconds

**HiFiBerryOS version**
hifiberryos-20210531

**HiFiBerry sound card**
HiFiBerry DAC+ ADC Pro

**To Reproduce**
Steps to reproduce the behavior:
1. Power On
2. Insert (input) 3.5mm input from media device
3. Insert (output) RCA to speakers or AMP, or receiver
4. Sound or Skip Loss every 6 - 7 Seconds
5. Rebooted several times - Same issue

**Expected behavior**
No skip loss

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Browser (if applicable)**
 - OS: N/A 
 - Browser [Chromium + Firefox]
 - Version 92.0.4515.107 (Official Build) snap (64-bit)
 - 90.0 (64-bit)

**Additional context**
N/A

hifiberry, 2021-08-18:
Unfortunately we can't reproduce this here. Please check the output of 

journalctl -f 

while this problem happens and post it here.

hifiberry, 2021-09-09:
Any feedback? Without logs, we can't do anything.

hifiberry, 2021-11-02:
As there was no feedback, I'm gonna close this.

closed 2021-11-02 by hifiberry