hifiberry/hifiberry-os #279: Pause-all can't pause snapcast (snapclient)

Bertware, 2021-07-27:
**Describe the bug**
When snapclient should be paused, the pause-all process fails

**HiFiBerryOS version**
20210531, previous releases had the bug as well

**HiFiBerry sound card**
DAC+ADC Pro

**To Reproduce**
1. Enable Alsaloop (Analog input) and snapcast
2. Play music through the analog input
3. ´20210728-003914 20210728-003924 pause-all alsaloop failed, snapclien still running´
4. Alsaloop doesn't start playing since the output device is occupied

**Expected behavior**
Snapcast/snapclient gets stopped, Alsaloop starts playing

**Additional context**
See https://github.com/hifiberry/hifiberry-os/blob/development/buildroot/package/hifiberry-tools/pause-all#L95, which defines "snapcast" as a possible name, but not "snapclien"
Call originates from https://github.com/hifiberry/alsaloop/blob/master/alsaloop.py#L181

ewlumpkin, 2021-08-11:
It looks like it keeps respawning this process even after it's killed.

hifiberry, 2022-01-13:
Probably fixed with https://github.com/hifiberry/hifiberry-os/issues/312

closed 2022-01-13 by hifiberry