hifiberry/hifiberry-os #280: Help Compilation

ssalaues, 2021-08-04:
**Describe the bug**
I would like to contribute to this project but I'm having difficultly running the `compile` script to build an image and test possible changes. Currently I'm trying to just build the baseline image and the only diff I have from the `master` branch is what seems to be a typo (listed below) in the `To Reproduce` section. I haven't been able to get past this stage so any help would be much appreciated.

**HiFiBerryOS version**
20210531

**HiFiBerry sound card**
DAC+ ADC

**To Reproduce**
Steps to reproduce the behavior:
1. Run `./prepare-software`
2. Fix missing quote in `Config.in` [here](https://github.com/hifiberry/hifiberry-os/blob/master/buildroot/Config.in#L47)
3. Run `./compile 3` (wait like 30 minutes)
4. See error

**Expected behavior**
Compilation of all sources prior to image building

**Screenshots**
```
mv: cannot stat '/mnt/c/Users/salim/github/buildroot-3/build/glibc-2.32-37-g760e1d287825fa91d4d5a0cc921340c740d803e2/build/nptl/stamp.oST': No such file or directory
make[3]: *** [../o-iterator.mk:9: /mnt/c/Users/salim/github/buildroot-3/build/glibc-2.32-37-g760e1d287825fa91d4d5a0cc921340c740d803e2/build/nptl/stamp.oS] Error 1
make[3]: *** Waiting for unfinished jobs....
make[3]: Leaving directory '/mnt/c/Users/salim/github/buildroot-3/build/glibc-2.32-37-g760e1d287825fa91d4d5a0cc921340c740d803e2/nptl'
make[2]: *** [Makefile:470: nptl/subdir_lib] Error 2
make[2]: Leaving directory '/mnt/c/Users/salim/github/buildroot-3/build/glibc-2.32-37-g760e1d287825fa91d4d5a0cc921340c740d803e2'
make[1]: *** [Makefile:9: all] Error 2
make[1]: Leaving directory '/mnt/c/Users/salim/github/buildroot-3/build/glibc-2.32-37-g760e1d287825fa91d4d5a0cc921340c740d803e2/build'
make: *** [package/pkg-generic.mk:250: /mnt/c/Users/salim/github/buildroot-3/build/glibc-2.32-37-g760e1d287825fa91d4d5a0cc921340c740d803e2/.stamp_built] Error 2
```

**Browser (if applicable)**
 - OS: Ubuntu
 - Browser [e.g. chrome, safari]
 - Version WSL2

**Additional context**
My end goal is to eventually integrate snapcast server (which seems to be just disabled in the buildroot config options). Since I can't really get the build working, I setup a [workaround ](https://github.com/ssalaues/hifiberry-snapserver)using the available docker interface and a fork of `alsaloop` to pipe the ADC input to the snapserver.

hifiberry, 2021-08-04:
You might not use the right buildroot version. Make sure you retrieve the correct version using 
get-buildroot

Also make sure, you have enough disk space. Compiling this needs A LOT of disk space.

ssalaues, 2021-08-05:
Thank you for the quick reply! Freeing up disk space definitely solved my first problem.

For anyone else having trouble setting up a dev environment:
- be sure to have enough free disk space!
- When using WSL be sure that the `hifiberry-os` git folder is **NOT** in the Windows filesystem because Windows does not differentiate between upper and lower case in file names causing many files to be overwritten and missing
- `./clean` is your friend

closed 2021-08-05 by ssalaues