hifiberry/hifiberry-os #282: AudioController/Power Controller: Button not working / pigpiod not triggering interrupt

j4n, 2021-08-17:
**Describe the bug**

Rotary Button on Power Controller does not work; looking into the problem it seems that the necessary pin change interrupt is not triggered:

`# journalctl -f | grep powercontroller &`
`# systemctl restart audiocontrol2 `
`# Aug 16 17:08:00 edafon python3[20716]: INFO: powercontroller - Using controller int pin 1 on GPIO 4`
`Aug 16 17:08:00 edafon python3[20716]: INFO: powercontroller - found powercontroller software version 5 on I2C address 119`
`Aug 16 17:08:00 edafon python3[20716]: INFO: audiocontrol2 - started controller powercontroller`
`Aug 16 17:08:00 edafon python3[20716]: DEBUG: connectionpool - http://127.0.0.1:3141 "POST /api/activate/audiocontrol_controller_ac2.plugins.control.powercontroller.Powercontroller HTTP/1.1" 200 2`
`Aug 16 17:08:00 edafon python3[20716]: INFO: powercontroller - Update LED state for state=unknown`

The pin registers the change though:

`# cat gpiodump`
`#!/usr/bin/env python3`
`import argparse, pigpio, time`
``
`parser = argparse.ArgumentParser("debug gpios")`
`parser.add_argument("gpio", type=int, help="GPIO Number")`
`parser.add_argument("--delay", type=float, default="0.01", help="delay between prints")args=parser.parse_args()`
``
`pi = pigpio.pi()`
`while True:`
` print(pi.read(args.gpio))`
` time.sleep(args.delay)`
`# ./gpiodump 4`
`1`
`1`
`0`
`0`
`1`
1`

When starting the pigpiod manually, interrupts worked at some point, but sound is slowed down; using the RPi gpio framework for Python, the interupts are also registered.

**HiFiBerryOS version**
20210804

**HiFiBerry sound card**
Beocreate4 w/ [40pin duplicator](https://www.reichelt.de/de/de/raspberry-pi-gpio-header-1-auf-2-40-polig-rm-2-54-rpi-gpio-1to2-p276993.html); reported make setup work in forum.

**To Reproduce**
1. Flash Power controller, install on beocreate
2. try rotating button to change volume

**Expected behavior**
Rotary encoder should register input and change volume/pause music/shut down OS.

**Additional Information**
[Forum thread](https://support.hifiberry.com/hc/en-us/community/posts/360015147497-Power-controller-beta-test?page=3)

hifiberry, 2021-09-09:
Trying to replicate this and it seems this is a Pi4 issue. Are you using a Pi4?

j4n, 2021-09-09:
Yes, this is a Pi4.

hifiberry, 2021-09-17:
I have a new experimental release ready with the bug fix. You might try it using 

/opt/hifiberry/bin/update --experimental

closed 2021-11-08 by hifiberry