hifiberry/hifiberry-os #283: OpenHome renderer not started properly

droebbel, 2021-08-17:
**Describe the bug**
On boot the OpenHome renderer is not started properly. After deactivating and reactivating it or restarting upmpdcli it reappears.

**HiFiBerryOS version**
20210804

**HiFiBerry sound card**
Beocreate

**To Reproduce**
1. Activate OpenHome renderer in the web interface
2. Check it is up and running with a control point like BubbleDS
3. Restart hifiberryOS
4. Check again - the renderer is missing
5. deactivate OH
6. activate OH
7. Check again - the renderer is back.

Alternative to 5./6.: ssh, # systemctl restart upmpdcli

**Expected behavior**
The renderer should be active after a reboot

I checked the /etc/upmpdcli.conf after a reboot - it looked correct, 
# Enable OpenHome services (0/1).
openhome = 1
was there.

hifiberry, 2022-10-10:
Is this still an active bug in 20220826?

droebbel, 2022-10-10:
Seems ok (I have not used the device for a while and cannot check right now, but as I forgot this issue completely I think it must be solved. Quite sure I rebootet the device several times without problems).
Sorry for not updating this!

hifiberry, 2022-10-10:
Thanks for the feedback.

closed 2022-10-10 by hifiberry