hifiberry/hifiberry-os #285: Room Correction, no Sweep

Jacco192, 2021-08-18:
DO NOT remove the blocks below, but fill these with the requested data. Incomplete bug reports will be ignored!
You should remove this comment before posting the bug report.

**Describe the bug**
The room correction option doesn't work as expected. When I start the room correction option and connect the microphone it is recognized and the room measurement starts, during the recording period there is silence and no sweep. I was playing around with the room correction and managed to get just one sweep, 1 out of 10 trials. How many sweeps should be expected during a recording ?
 
HifiBerryOS, PI 4, Digi2 Pro, DSP

**HiFiBerryOS version**
20210804

**HiFiBerry sound card**
Digi2 Pro with DSP

**To Reproduce**
Start room correction.

**Expected behavior**
Sweep to record for setting the room correction

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Browser (if applicable)**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

hifiberry, 2021-08-18:
Try the following:

- login via SSH
- run 
  /opt/hifiberry/bin/run-measurements
  
If this still doesn't output a sweep, run

/opt/hifiberry/bin/pause-all

and run the 

 /opt/hifiberry/bin/run-measurements

again

Jacco192, 2021-08-18:
Solved, /opt/hifiberry/bin/pause-all did the trick.
Running room compensation from the GUI gives me now 4 sweeps, nice.
Thanks !

hifiberry, 2021-08-18:
Ok, we'll include this into the measurement script

hifiberry, 2021-11-08:
Fixed in the new release.

closed 2021-11-08 by hifiberry