hifiberry/hifiberry-os #288: Bluetooth Playback Volume control and pairing broken on Rpi4 in 20210804

glockman, 2021-09-03:
**Describe the bug**
- Pairing does not work (neither under Windows, nor under Android)
- If connected based on a formerly established pairing, playback works, but volume control does not work

**HiFiBerryOS version**
20210804 on Raspberry Pi4

**HiFiBerry sound card**
DAC+ DSP

**To Reproduce**
- revert to 20210531
- pair the hifiberry
- update to 20210804
- see bluetooth connection and playback working, but volume control not working (at leat in windows)
- see pairing not working despite the hifiberry appearing in the available bluetooth devices / scan

**Expected behavior**
A working pairing process and volume control.

**Screenshots**
N/A

**Browser (if applicable)**
N/A

**Additional context**
While my windows 10 notebook could still connect and play sounds after the automatic update to 20210804, the volume control didn't work. Restarting the Hifiberry and even restarting my notebook (Intel BT) did not change the situation.
Unpairing the Hifiberry under Windows and trying to pair again didn't work. In the scan the Hifiberry pops up but pairing fails.
From my Android Phone pairing didn't work either. 
Turning Bluetooth off and on again on the  doesn't make a difference.
Reverting to 20210531 does lead to the expected behaviour again.

glockman, 2021-09-03:
I'm not alone with this behaviour. See https://support.hifiberry.com/hc/en-us/community/posts/4405558354833-Bluetooth-no-longer-working-after-last-software-update-DAC-DSP-with-Raspberry-Pi-4-

With the thread opener, my upvote and another ones upvote it seems that at least 3 people are seeing this behaviour.

cfhsu, 2021-09-09:
I have had the same bluetooth problem with my RPi 4B and software 20210804 as well. Finally got it solved by logging in SSH and using the bluetoothctl command to delete my paired devices. Afterwards the device can be pared again.

hifiberry, 2021-09-09:
Interesting. Might be due to an updated version of the Bluetooth stack

@glockman : Can you try this?

glockman, 2021-09-09:
Removing the paired-devices indeed fixed the pairing problem. 
So for an average user to be able to help themselves a list with the paired-devices in the web-ui along with the possibility to remove them would be of great help in such situations.

But the volume control does not work at all with Windows. And with Android BT-audio volume is not coupled with HifiberryOS system volume like it was with 20210531.

glockman, 2021-09-09:
I wouldn't mind the volume control not being synced with bluetooth audio, this would even fit my setup far better. But no volume control from windows at all is problematic as 2x 50W Monitor-Speakers on the desk at full volume can range from very surprising to quite painful.

My work notebook is connected via bluetooth and my private desktop connected via SPDIF. So I often find myself having to use the webUI to increase the volume back up when I lowered it during work.

hifiberry, 2021-09-09:
To be clear here: BT is the least robust player in HBOS and as long as there isn't anybody with deep BT know how that wants to contribute, it will probably be like this also in the future. BT implementations vary widely and we don't have the hardware and time to test with hundreds of devices. While this might improve, it's probably best for you (at least for now) to stick with a release that works best for you.

hifiberry, 2021-09-14:
Building a new release with a newer version of the BT audio stack now. Works here, but I only have an iPhone.

Steinarhem, 2021-09-15:
I have the same issue on Beocreate 4Ch

hifiberry, 2021-09-17:
I have a new experimental release ready with some changes that might fix this. You might try it using 

/opt/hifiberry/bin/update --experimental

Steinarhem, 2021-09-19:
I tested this experimental now. Volume did not work on Android bluetooth

Steinarhem, 2021-09-19:
I can't go back to 0210203 with the "Previous version" after testing the experimental.

Is it a way of doing the revert? I'd like a functioning volume

hifiberry, 2021-09-19:
What do you mean by "can't go back". Reverting to the previous version should 
always work.

Steinarhem, 2021-09-19:
I upgraded first to the x804 version, then I ran the experimental you posted.
Now I can't go back.

[image]

pinballpower, 2021-09-19:
You can only go back to the previous release. You can find older releases here:
https://github.com/hifiberry/hifiberry-os/releases

derhagen, 2021-09-23:
> I have a new experimental release ready with some changes that might fix this. You might try it using
> 
> /opt/hifiberry/bin/update --experimental

Volume control not working with Fedora 34 / RPi4 on latest HBOS. Experimental update did not fix it.

hifiberry, 2021-09-23:
What do you mean by Fedora? Do you run Fedora or HBOS?

derhagen, 2021-09-23:
I'll open a pull request in a sec

hifiberry, 2021-09-27:
This seems to be a problem that needs to be address in bluealsa as it seems that volume information isn't always correctly forwarded.

hifiberry, 2022-10-10:
As this bug has been closed in bluez, we'll close it also here. Upgrading to Android 11 or later should fix this.

closed 2022-10-10 by hifiberry