hifiberry/hifiberry-os #290: Radio is restarted automatically after it has been stopped

yubiuser, 2021-09-09:
**Describe the bug**
There is no way to delete the library.

**HiFiBerryOS version**
20210804
**HiFiBerry sound card**
"DAC+ADC".

**To Reproduce**
Steps to reproduce the behavior:
1. Start playing a radio stream
2. Click on `Stop`
3. See how Music is restarted a second later

**Expected behavior**
After stopping, music should stay off.

**Additional context**
It doesn't happen every time but has happend a few times now. I'm not sure how to reproduce it reliable. 

[image]

hifiberry, 2021-09-14:
This is quite interesting. Unfortunately I can't reproduce it here. What Pi are you using?

yubiuser, 2021-09-14:
It's a Pi4

hifiberry, 2021-09-14:
Interesting. What other players are enabled?

yubiuser, 2021-09-14:
Bluetooth, Airpaly, Spotify, Music, Radio

hifiberry, 2021-09-14:
Try disabling the other sources and see if this changes something - especially Bluetooth.

yubiuser, 2021-09-14:
Ok, I'll test a bit

hifiberry, 2021-11-04:
Any feedback on this?

hazet42, 2022-01-04:
Same problem here - only that I am using HifiBerry OS on a raspberry Pi 3 with HifiBerry DAC+ ADC. Sources enabled are Music, OpenHome and Radio. Can I do something to help with this issue?

yubiuser, 2022-08-04:
I could not reproduce the issue with the latest version.

closed 2022-08-04 by yubiuser