hifiberry/hifiberry-os #292: Bluetooth is >=250ms delayed, disturbs lip sync

derhagen, 2021-09-23:
DO NOT remove the blocks below, but fill these with the requested data. Incomplete bug reports will be ignored!
You should remove this comment before posting the bug report.

**Describe the bug**
Bluetooth audio is at least 250ms delayed. That's almost unbearable when watching a movie.

**HiFiBerryOS version**
Latest experimental

**HiFiBerry sound card**
HiFiBerry DAC+ Zero

**To Reproduce**
Just play audio over bluetooth

**Expected behavior**
There should be no noticable delay when watching a movie

hifiberry, 2021-09-27:
This is not a bug, but a feature request.

derhagen, 2021-09-27:
Pretty much every single bluetooth speaker (including the cheapes of aliexpress bt-amps) on the market comes with almost perfectly synchronous audio. So yes, I'd consider this a bug. If you bought a TV and it only had b/w image, you wouldn't open a feature request and ask for colors, you would return it.

thefiddler, 2021-12-05:
@derhagen all bluetooth speakers have non-zero delay. For example, my Sony MDR-100ABN headphones have roughly 250ms of delay on windows and macos (linux is comparable but I didn't try to measure it.)

While it would be interesting to have additional controls in HifiBerryOS, some latency is unavoidable due to how bluetooth works.

krnlde, 2021-12-05:
Why are Bose headsets and the Jabra stations completely in sync with Teams then. Someone is lying.

derhagen, 2021-12-05:
@thefiddler "no noticable delay" and "zero delay" are different things. I don't think this sort of discussion is not very productive and I won't engage in it any more, since I already clarified my requirements / preferences or however you want to call it.

I proposed a fix in the yet unmerged PR https://github.com/hifiberry/hifiberry-os/pull/293 that reduces the buffer to barely noticeable 20ms. It works perfectly fine for me. @hifiberry claimed that "It's not a recommendation to set the buffer time o 20ms." But as long as they don't say who gave this recommendation and why, I can't argue against it.

dmnc-net, 2021-12-13:
Even 400ms delay is normal issue with BT device, depends on hardware and additional workarounds. Hotfix tip: depends on used OS but PulseAudio/PipeWire supports 'latecy offset', see `pavucontrol` (GUI) or `pactl set-port-latency-offset` (CLI).

closed 2022-07-21 by hifiberry

hifiberry, 2022-07-21:
No. As nobody is working on this, it has been closed.

hifiberry, 2022-07-21:
I answered already in your pull request:

"It's not a recommendation to set the buffer time o 20ms. As small buffers might result in potential problems, we won't accept this as a default setting.
Feel free to provide a PULL request that allows users to change a shorter buffer time without making it a default setting for all users."

While you might not agree with this, this is our position and we will only accept a change that doesn't change the default setting.