hifiberry/hifiberry-os #294: No cover art in when HiFiBerryOS play as DLNA renderer

ayodia1, 2021-09-26:
**Describe the bug**
When HifiBerryOS play music as DLNA render, from mobile phone (for example, bubble upnp) the cover art is not displayed. 

**HiFiBerryOS version**
20210804

**HiFiBerry sound card**
DAC+

**To Reproduce**
Steps to reproduce the behavior:
1. select HiFiBerry as renderer in DLNA supported app in mobile
2. Play any music from mobile app
3. The music is played with song name, but no cover art

**Expected behavior**
Display the cover art as the music file include it.

**Screenshots**

**Browser (if applicable)**

**Additional context**
I tried Moode and other DLNA render - I could see the cover art correctly. 
I tried other music app (Melon) in mobile - HiFiBerryOS does not display the cover art.

hifiberry, 2021-09-26:
It's not a bug, but a feature request.

ayodia1, 2021-09-27:
I see. But, please note, there are many music streaming services in the world and I don't think HiFiBerryOS cannot accommodate every service natively. Most of music streaming service provide DLNA audio out. So, I think this is important use case.

hifiberry, 2021-09-27:
According to the usage of the services, there are very few DLNA users today. Therefore, this has a very low priority for us. However, we're open for contributions.