hifiberry/hifiberry-os #295: No Korean/Japanese character display in RaspberryPi HDMI screen

ayodia1, 2021-09-26:
**Describe the bug**
When I connect the HiFiBerryOS from PC browser, I can see Korean characters but it is not correctly displayed in Raspberry Pi HDMI screen (Maybe font is missing?)

**HiFiBerryOS version**
20210804

**HiFiBerry sound card**
DAC+ Pro

**To Reproduce**
Steps to reproduce the behavior:
1. Select HiFiBerryOS as DLNA render from mobile app
2. Play any Korean music with Korean title
3. The Korean character is not displayed correctly in Raspberry Pi HDMI screen

**Expected behavior**
Display correct characters 

**Screenshots**

**Browser (if applicable)**

**Additional context**
I tried Japanese as well. It is same. I can see correct characters in PC browser, but problem in Raspberry Pi HDMI display.

hifiberry, 2021-09-27:
Probably missing font. I'm not sure, if we can get the correct fonts, but I'll have a look. As I don't have any files to test: Can you provide a sample file?

ayodia1, 2021-09-27:
There is possible solution. I found moode audio also support HDMI UI out and they have same problem. But I could add the fonts by below (they have web SSH) and I could see the characters properly. 

$ sudo apt install -y fonts-unfonts-core

Here is the sample song with Korean character in metadata - I attached the screenshot of Korean string, so that you can know if it is correctly displayed or not

[mp3.zip](https://github.com/hifiberry/hifiberry-os/files/7235902/mp3.zip)

[image]

hifiberry, 2024-04-10:
63564ffd65cf6a752f518e637bedad4e9a58c1c7

Try the upcoming HBOS alpha 7