hifiberry/hifiberry-os #299: Not working new (20211105) version on https://www.hifiberry.com/hifiberryos/

HansSchukkink, 2021-11-07:
**Describe the bug**
There is a new version on https://www.hifiberry.com/images/hifiberryos-pi3.zip.
After unpacking this results in hifiberryos-20211105-pi3.img
This is NEWER than 'latest' on https://github.com/hifiberry/hifiberry-os/releases
And the problem is that it does not boot. (3B+ with AMP+), 32GB microSD.

**HiFiBerryOS version**
20211105

**HiFiBerry sound card**
AMP+

**To Reproduce**
Steps to reproduce the behavior:
Flash the image and try to boot.

**Expected behavior**
I expect it to boot.

**Screenshots**
N/A

**Browser (if applicable)**
N/A

**Additional context**
N/A

HansSchukkink, 2021-11-07:
Actual boot problem was likely due to defect SD card. Only minor problem is that the version is not documented on github.

pinballpower, 2021-11-08:
We usually test the latest experimental with some users before pushing it as the "latest" version to the updater and to Github. This is the reason for this newer release on our web site. We expect this to become the "latest" version soon.

hifiberry, 2021-11-08:
We usually test the latest experimental with some users before pushing it as the "latest" version to the updater and to Github. This is the reason for this newer release on our web site. We expect this to become the "latest" version soon.

closed 2021-11-08 by hifiberry