hifiberry/hifiberry-os #301: How do I build the OS? Error: Your old configuration uses legacy options

theblond, 2021-11-10:
**Describe the bug**
I try to build a Hifiberry OS image, but I got an error message that I use an old configuration with legacy options.

**To Reproduce**
I downloaded the git repository of Buildroot and Hifiberry OS and followed the steps described here: [building.md](https://github.com/hifiberry/hifiberry-os/blob/master/doc/building.md)
In detail I did:
`./build config 4` (that worked)
`.compile 4` (this does not)

**Expected behavior**
Is the standard configuration here on Github not meant to work?

**Screenshots**
[image]
[image]

hifiberry, 2021-11-10:
You need to use the get-buildroot script to download and patch the correct buildroot version.

theblond, 2021-11-10:
got it, thank you!

closed 2021-11-10 by theblond