hifiberry/hifiberry-os #302: Version 20211105 misdetects DAC card

DaleEMartin, 2021-11-11:
I upgraded to 20211105 and now it appears as though AirPlay connects and stays connected but there is no sound. I tried Bluetooth as a workaround and there is no sound using Bluetooth either.

Looking at the Diagnostics page it says “DAC+ Zero/Light/MiniAmp” for “Sound Card” which I don’t believe is correct. I have a DAC2 HD. Sorry for the delay in testing/debugging this.

hifiberry, 2021-11-11:
Try adding the correct overlay to /boot/config.txt and check if this helps. If it doesn't, there is some kind of hardware problem.

DaleEMartin, 2021-11-11:
It worked. Here were the steps for anyone else trying this. Logged in as root:
mount -o remount,rw /boot
Then change the contents of /boot/config.txt from:
dtoverlay=hifiberry-blocked
dtoverlay=hifiberry-dac

To:
dtoverlay=hifiberry-blocked
#dtoverlay=hifiberry-dac
dtoverlay=hifiberry-dacplushd

Do “sync” to make sure the FS gets flushed and then reboot.

I’ve never had to touch any of this stuff before. I’m glad it’s working now though, thanks!

hifiberry, 2021-11-11:
Hmm, are you using the DSP add-on?

DaleEMartin, 2021-11-11:
No

DaleEMartin, 2021-11-11:
Here you go: 
https://www.icloud.com/iclouddrive/0srR1lrOo3ArQI_2O9DwihcXA#hifiberry-debug

That’s the debug info from my system

hifiberry, 2021-11-11:
Unfortunately that doesn't help. We haven't seen this behavior before. We need to monitor if other users see the same problem.

DaleEMartin, 2021-11-11:
Ok. Happy to do a little more debugging or add more info if you think of something to try. I think my setup is pretty vanilla.

JCBird1012, 2022-04-24:
I see this behavior as well - running DAC+ Pro and for some reason, every time I modify /boot/config.txt, it gets reset back to it's previous value. It doesn't persist between reboots.

JCBird1012, 2022-04-24:
After some further tinkering, including installing vanilla Raspberry Pi OS and trying the DAC+ Pro on another Pi - I've come to the somewhat plausible conclusion that my DAC+ Pro is now dead - I'm not too too upset - this DAC+ Pro has served me for two years and has been through its paces - so disregard my previous comment.

Once I get another (hopefully) working DAC+ Pro, I'll report back to see if I'm still experiencing the issue.

Uzarel, 2022-05-14:
> I see this behavior as well - running DAC+ Pro and for some reason, every time I modify /boot/config.txt, it gets reset back to it's previous value. It doesn't persist between reboots.

Same here, even if my DAC is just 1 month old!

closed 2022-05-17 by hifiberry