hifiberry/hifiberry-os #303: Beocreate 4 channel amplifier incorrectly detected as DAC+ DSP in 20211105

thefiddler, 2021-11-12:
**Describe the bug**
Beocreate 4 channel amplifier is incorrectly detected as DAC+ DSP. No sound.

**HiFiBerryOS version**
20211105

**HiFiBerry sound card**
Beocreate 4 channel amplifier

**To Reproduce**
Steps to reproduce the behavior:
1. Install an AMP2 and configure it - works fine
2. Replace the AMP2 with a Beocreate - appears as DAC+ DSP instead
3. Try to play any sound (airplay or bluetooth) - no sound

**Expected behavior**
Beocreate 4 channel amplifier should be detected and used.

**Screenshots**
[image]

**Browser (if applicable)**
 - OS: MacOS
 - Browser: firefox
 - Version: 94

**Additional context**
I was successfully using AMP2 before.

After installing the Beocreate, the DSP program was listed as **Unknown Program**.

I followed the instructions from Norbert here https://support.hifiberry.com/hc/en-us/community/posts/4409631891217-Improved-DAC-DSP-Beocreate-detection:
- `dsptoolkit install-profile /opt/beocreate/beo-dsp-programs/beocreate-universal-10.xml` made the DSP program appear as Beocreate Universal (see screenshot below)
- `nano /etc/hifiberry/config.fixed` made the Beocreate be detected
- however still no sound out of the speakers. (Note: I have removed the mute jumper, so that's not it)

[image]

thefiddler, 2021-11-12:
If I run `/opt/hifiberry/bin/reconfigure-players` **without** a `/etc/hifiberry/config.fixed` I get the wrong sound card (DAC+ DSP):
```
/opt/hifiberry/bin/readhat: line 23: /sys/class/i2c-adapter/i2c-/new_device: No such file or directory
/etc/hifiberry ~
~
Feature pi not yet configured
Feature dsp not yet configured
Sound card settings have been changed, reconfiguring...
Applying settings:
 Players:  (was )
 Raspberry Pi model: 4
 System name: egli-hifi (was egli-hifi)
 Sound card: DAC+ DSP (was Beocreate 4-Channel Amplifier)
 Interrupt pin for power controller: 2 (was )
 DSP type: DAC+ DSP (was )
 HW device: card 0 (was card 0)
 ALSA mixer control: DSPVolume (was DSPVolume)
 Recording:  (was )
 Vendor/product   (was )
 Features: pi bluetooth pi3orlater arm7 dsp toslink localui (was !)

Saving settings to /etc/hifiberry.state
Reconfigure features
Reconfiguring Spotifyd
Reconfiguring Vollibrespot
Reconfiguring Shairport-sync
Reconfiguring Bluetooth
Reconfiguring squeezelite
Reconfiguring MPD
Reconfiguring DLNA
Reconfiguring Mopidy
modidy not installed, ignoring
Reconfigure AudioControl2
 updated interrupt pin
Restarting audiocontrol2
Reconfigure upmpdcli
Reconfigure RAAT
Configuring RAAT server for UUID 78142091-3786-441b-9c55-94a1a9e9e085 on DAC+ (snd_rpi_hifiberry_dac)
Reconfiguring Beocreate software
Value could not be interpreted as JSON, falling back to old method.
Settings updated for 'system'.
 updated card type
["pi","bluetooth","pi3orlater","arm7","dsp","toslink","localui"]
Settings updated for 'system'.
Value could not be interpreted as JSON, falling back to old method.
Settings updated for 'system'.
Restarting beocreate2
 restarting Beocreate
```

If I run `/opt/hifiberry/bin/reconfigure-players` **with** a `/etc/hifiberry/config.fixed` I get the right Beocreate sound card, but the DSP disappears:
```
# /etc/hifiberry/config.fixed
MIXER_CONTROL="DSPVolume"
CARD="Beocreate 4-Channel Amplifier"
FEATURES="toslink localui"
```
```
/opt/hifiberry/bin/reconfigure-players: line 50: $N=${CURRENT_$N}: bad substitution
Using fixed card Beocreate 4-Channel Amplifier with fetaures toslink localui bluetooth pi3orlater arm7
Using fixed mixer control DSPVolume
/etc/hifiberry ~
~
Sound card settings have been changed, reconfiguring...
Applying settings:
 Players:  (was )
 Raspberry Pi model: 4
 System name: egli-hifi (was egli-hifi)
 Sound card: Beocreate 4-Channel Amplifier (was DAC+ DSP)
 Interrupt pin for power controller:  (was 2)
 DSP type:  (was DAC+ DSP)
 HW device: card 0 (was card 0)
 ALSA mixer control: DSPVolume (was DSPVolume)
 Recording:  (was )
 Vendor/product  (was  )
 Features: toslink localui bluetooth pi3orlater arm7 localui (was pi bluetooth pi3orlater arm7 dsp toslink localui)

Saving settings to /etc/hifiberry.state
Reconfigure features
Reconfiguring Spotifyd
Reconfiguring Vollibrespot
Reconfiguring Shairport-sync
Reconfiguring Bluetooth
Reconfiguring squeezelite
Reconfiguring MPD
Reconfiguring DLNA
Reconfiguring Mopidy
modidy not installed, ignoring
Reconfigure AudioControl2
 updated interrupt pin
Restarting audiocontrol2
Reconfigure upmpdcli
Reconfigure RAAT
Configuring RAAT server for UUID 78142091-3786-441b-9c55-94a1a9e9e085 on DAC+ (snd_rpi_hifiberry_dac)
Reconfiguring Beocreate software
Value could not be interpreted as JSON, falling back to old method.
Settings updated for 'system'.
 updated card type
["toslink","localui","bluetooth","pi3orlater","arm7","localui"]
Settings updated for 'system'.
Restarting beocreate2
 restarting Beocreate
```

hifiberry, 2021-11-16:
Can you please post the output of 

/opt/hifiberry/bin/read-hat

thefiddler, 2021-12-05:
Running `/opt/hifiberry/bin/readhat` prints:
```
/opt/hifiberry/bin/readhat: line 23: /sys/class/i2c-adapter/i2c-/new_device: No such file or directory
```

hifiberry, 2022-08-04:
This should have been fixed some time ago. Can you test with the latest release?

closed 2022-10-10 by hifiberry

jedie, 2022-10-20:
Think i have the same Problem with current 20220826 release!

[image]

```
# /opt/hifiberry/bin/readhat
no vendor:no product
```
I tried without and with `/etc/hifiberry/config.fixed` with:
```
MIXER_CONTROL="DSPVolume"
CARD="Beocreate 4-Channel Amplifier"
FEATURES="toslink localui dsp"
```

[image]