hifiberry/hifiberry-os #306: mac address error for squeezelite

jamesholdenmusic, 2021-11-17:
DO NOT remove the blocks below, but fill these with the requested data. Incomplete bug reports will be ignored!
You should remove this comment before posting the bug report.

**Describe the bug**
OS needs to wait for network connection before starting squeezelite, as otherwise squeezelite reports an all-zeroes mac address (which leads to conflicts with multiple players as the logitech media server software identifies players by mac address)

**HiFiBerryOS version**
all

**HiFiBerry sound card**
all

**To Reproduce**
Steps to reproduce the behavior:
1. set up multiple squeezelite / hifiberry players on your network
2. look in LMS player list, they all have 00:00:00:etc mac addresses, and hence cannot coexist simultaneously

**Expected behavior**
hifiberry os should wait before starting squeezelite, then the squeezelite player will report a correct mac address to the server

**Screenshots**
n/a

**Browser (if applicable)**
n/a

**Additional context**

hifiberry, 2022-01-13:
Addresses with 9f53175178c9da83d3aca9afc29f74ffc53fa0bf

hifiberry, 2022-01-25:
This has been implemented in release 20220114

closed 2022-01-25 by hifiberry