hifiberry/hifiberry-os #307: Browse by directoryi

williemctell, 2021-11-22:
I've installed HifiBerryOS and just discovered that you can only access music by tag. I have music files dating back over 20 years. They're tagged haphazardly if at all. Most of them are ripped CDs and LPs . I stored them by directory because it makes more sense to me than tags. At this point with more than 80K files it would take me an unreasonable amount of time to tag the whole collection uniformly. I understand that there is some way to activate ympd, bundled with the software, but systemctl --now enable ympd seems to do nothing that shows up in the GUI.

Every other MPD front end I've seen provides for browsing by directory, including Moode, Volumio, Kodi, and the applications for mobile devices and Linux. I'd appreciate seeing directory based browsing.

hifiberry, 2021-11-22:
ympd runs on port 9999. There is no integration in the HiFiBerryOS UI and we don't plan this. 
We might look into contributions from other developers, but we don't think that browsing by directoy is a good way to interact with the music library. 
I strongly recommend to spend some time to tag your files correctly. You might have a look here:
https://www.hifiberry.com/blog/tools-to-tag-your-music-collection/

closed 2021-11-22 by hifiberry

williemctell, 2021-11-22:
Umm...Your response was a bit judgemental.  Since mpd supports organization by directory it doesn't seem like an unreasonable request for an mpd front end.   

There is no such thing as tagging files correctly.  Tag fields are text fields that can contain anything.  A basic rule of database design is that when end users are given complete control over a field it will contain anything.  There is no guarantee that tags will be consistent from source to source.  I have programming experience with user controlled attributes.  Given the large number of files I have, even an automated retag of the entire collection would take an unreasonable amount of time.  

BTW, organization by directory is a very reasonable method of file organization.  From my point of view it's the easiest and most convenient.  It allows me to find anything I want quickly.  Admittedly I am responsible for maintaining the consistency of the system but that's the norm for filing of any sort--paper, electronic,...  Tagging, OTOH requires depending on the kindness of strangers.

hifiberry, 2021-11-22:
We don't think "files" and "directories" are the right way to experience music. I understand that you don't agree with this and I understand why do you want to us it differently. That's fine and we accept this.  However, we can't create a software that works for everybody. We have made a design choice and we are aware not everybody might like it.