hifiberry/hifiberry-os #308: Compiling Error: python keyboard failed to create device

theblond, 2021-11-22:
**Describe the bug**
I am getting the followring error message when compiling the standard Hifiberry-OS build.

> ERROR: Failed to read device '/dev/input/event0'. You must be in the 'input' group to access global events. Use 'sudo usermod -a -G input USERNAME' to add user to the required group.
> Couldn't get a file descriptor referring to the console  It always stops while compiling the python-keyboard package.

I tried the suggested commands in the error message  `sudo usermod -a -G input USERNAME` and `sudo useradd -a -G tty USER` without success.

**HiFiBerryOS version**
20211105

**To Reproduce**
Run the Hifiberry OS building process for Raspberry Pi 4 on Ubuntu 20.04 server.

**Expected behavior**
Runs without errors.

**Additional context**
[Building log](https://github.com/hifiberry/hifiberry-os/files/7584382/Python.keyboard.building.error.txt)

hifiberry, 2021-11-23:
Unfortunately I can't really help with this. The build on Ubuntu works just fine here. Are you sure, you're using the correct buildroot version?

theblond, 2021-11-23:
I did this on a clean install of Ubuntu:
> sudo apt update && sudo apt upgrade -y
> git clone https://github.com/hifiberry/hifiberry-os.git
> cd hifiberry-os
> ./prepare-software
> ./get-buildroot
> ./build-config 4
> ./compile 4

The only thing I changed is the prepare-software file, because it does not find the packages:
> ncurses-dev -> libncurses-dev
> python -> python3
> python-alsa -> python3-alsa

Buildroot version is 2021.05

hifiberry, 2021-11-24:
We're still using Ubuntu 18.05 LTS here. Are you sure, you were using the correct user in the usermod command?

theblond, 2021-12-03:
Well, I found the issue after a lot of headaches.

I wasn't part of the `input` and `tty` group. As it was mentioned in the error log, but I had not rebooted, so it didn't take effect...

To solve this issue I add my username to the two groups and reboot:
`sudo usermod -a -G tty,input USERNAME`

Other people had this issue with that keyboard package, too. The author gave the right hint: [link to comment](https://github.com/boppreh/keyboard/issues/312#issuecomment-929950601)

There is still a warning message, but I can now successfully build the Hifiberry OS image:
`Failed to create a device file using uinput module. Sending of events may be limited or unavailable depending on plugged-in devices.`

After that intense testing, i can confirm that one can build Hifiberry OS on Ubuntu 20.04 and Ubuntu 18.04 virtualized through Hyper-V generation 2 or WSL version 2.

closed 2021-12-03 by theblond